LumaTeam/Luma3DS #109: GBA titles don't work on emunand

iamapie13, 2016-06-04:
Hello, I just installed the latest version of luma in my new 3DS with menuhax, and all my GBA titles stopped working unless I use the Sysnand, but then I have tu unwrap all my games again in my emunand

urherenow, 2016-06-04:
Unwrap is because your nands are linked. If you have the same gba stuff in sysnand and they don't work from emunand, you've installed a patched agb_firm. Install the stock version.

Or... If that's not the case, boot Luma holding select and change your option to show gba bios thingy...

LinkofHyrule89, 2016-06-11:
I tried playing the DSi Ware of The Legend of Zelda Four Swords Aniversary Edition in Emunand and that didn't seem to work either.

cheatfreak47, 2016-06-11:
This problem is not a Luma 3DS problem, but instead a general CFW problem.
At this time we don't have the means to redirect reads/writes from SysNAND to EmuNAND while in firmware modes other than the standard CTR firmware mode, it may become possible in the future, however the need for it is not particularly prevalent, as most people now use Arm9LoaderHax and Luma 3DS with their SysNAND.

Because of this, it is needed to install both on your SysNAND and EmuNAND in order to play games that use TWL or AGB firmware modes, as when the system switches modes, it will then proceed to look for the needed content in whatever NAND the SysNAND is tied to, if the content is there, no problem, play the game, if not, it returns the standard nintendo console error screen.

It may be possible in the future with some research to allow GBA and DSiware from EmuNAND, but in that case, you will have to be using Arm9LoaderHax as your entry point, since the console has to reboot into those firmware modes, and other entrypoints don't run before AGB or TWL mode start, thus not giving us the chance to patch anything.

Basically, as it is right now, it is not possible to play GBA and DSiWare from an EmuNAND alone, it must be used in conjunction with the same content being installed on your SysNAND.

@iamapie13 Your issue with gifts is related to your EmuNAND and SysNAND being linked, IE They are using the same folders on the SD card and the data on the NAND is desynced, causing you to have to unwrap the games each time you encounter them, to fix this, format one of your NANDs with TinyFormat.

@LinkofHyrule89 The issue you are having is precisely the same issue as the original poster, you must install the DSiware to both your SysNAND and EmuNAND.

This issue is invalid and can be closed.

closed 2016-06-11 by RShadowhand

iamapie13, 2016-06-11:
I fix the problem I had with the GBA games crashing on emunand, not by any of rhis comments unfortunately, but by reinstalling Luma 3DS, and by the way, for some reason my GBA games can work by only been installed in emunand and not being installed in sysnand

iamapie13, 2016-06-11:
I don't use arm9loaderhax because of my 2 years warante

fincs, 2016-06-11:
Your warranty is already void due to the use of "unauthorized" software.

GlitchedCode, 2016-12-18:
arm9loaderhax and other unathorized software didn't void my warranty, i sent my 3ds in to repair and they worked it out, i just made sure to replace my sdcard contents with a backup from long ago when i still didn't have a cfw