LumaTeam/Luma3DS #112: System Goes To Error Screen After 5.4 - > 5.5 Update

Gemfruit, 2016-06-06:
Copying this from my reddit post, as I'm tired. All info below:

So I'm more than capable of restoring my NAND and getting my setup working again, so this is not a post asking for help, this is a post about figuring out what went wrong.

Details

a9lh
N3DSXL
64GB Samsung Evo

Pretty vanilla setup - I followed the guide, uncarted some games, and played said games. I have no funky CIAs installed, no other bells and whistles, just a completed a9lh setup from the guide, and my few games / emulators (Retroarch).

So two days ago, I used LumaUpdater to update my system to 5.5, because it was there, and why not. The system hung for about 60+ seconds trying to do this, and then completed. I powered it down, and didn't turn it back on for a day. When I did, it boots up, shows my home screen for ONE second, and then brings me to "An error has occurred" black screen of doom. Here's a video link showing this - https://dl.dropboxusercontent.com/u/2660581/2016-06-05%2023.38.04.mp4

I've tried removing luma/firmware.bin, renaming my Nintendo 3DS folder, replacing arm9loaderhax with 5.4, 5.5, etc. Nothing I do seems to work, and the #3dshacks irc chat is pretty stumped too.

Again, I can fix this, that isn't the issue, the issue is that I have no idea what went wrong, and it's burning a whole in my logical-filled brain. Anyone have any ideas?

nickthegamer5, 2016-06-06:
Can you boot decrypt9(or hourglass9) and restore from a backup? I think you bricked your 3DS

urherenow, 2016-06-06:
Not only remove any .bin from the luma folder (even the config), but also try renaming your Nintendo 3ds folder and see if it helps. If it doesn'thelp, the problem is with either something in nand, or an outdated arm9loaderhax (try installing the latest screen init version with safea9lh installer).

Gemfruit, 2016-06-06:
I have renamed the Nintendo 3DS folder, and replaced arm9loaderhax with 5.4, and 5.5. The dev build gave me this error:

http://i.imgur.com/GxP3Lfn.jpg

TuxSH, 2016-06-06:
I don't think it's Luma3DS's fault.

The error is related to head-tracking.

RShadowhand, 2016-06-06:
Not related to Luma 3DS, but it happened right after @Gemfruit used Luma updater. Might be a coincidence, but I think there's something to that.

urherenow, 2016-06-06:
Luma Updater is not part if Luma3ds, is it?

Gemfruit, 2016-06-06:
Story not included here, that I thought was unrelated, but makes sense given the error, is that I was drunk wrestling with the 3DS in my pocket that night (terrible decision). I don't recall hitting it hard, and there is zero visible damage, but my guess is the camera / tracking got dislodged, as I hear a tiny rattling that I cannot differentiate from the buttons (could be either). That being said, I'm going to give Luma Updater the benefit of the doubt, and assume it was coincidence. Given everything I've tried (including recovery mode, and NAND restore), combined with that error and the potential rattling noise, I'm going to have to assume I broke the camera drunk.

If anyone knows of any possible way to disable the head tracking on the software side so the system will ignore it and not error out, that would be amazing. Otherwise, I need to buy a new 3DS, and I'm boned.

Moral of the story, don't drink and 3DS.

nickthegamer5, 2016-06-06:
I would say send it in to get t fixed but Nintendo would not be happy when they see you modified the arm9  chip :3 but beside that unless you know how to replace the 3ds camera I think your out of luck :/

urherenow, 2016-06-06:
If you could find the old a9lh guide, there were instructions for injecting the o3ds headers to make 2.1 boot. If disabling head tracking could fix this issue, then this would do it. If you've already done a9lh and have a copy of the working sysnand, try to flash that one and see if you still get the error. As I said before though, even though flashing an older backup confirms that Luma updater didn't do this, this git isn't the right place anyway. Luma updater is a different project (unless I'm just blind... I'm on my phone only, until tomorrow evening)

TuxSH, 2016-06-06:
Could you try this build, and, if it works, disable head tracking?

https://dl.dropboxusercontent.com/u/57553042/out_qtm_stub.7z

closed 2016-06-12 by TuxSH