LumaTeam/Luma3DS #113: GW's arm9loaderhax.bin support on the chainloader for no-screeninit users.

billy-acuna, 2016-06-06:
Looks like GW's payload or launcher.dat (dunno at all) doesn't like no-screeninit A9LH setup.
Is this fixable by including screeninit code to "(botton)_GW.bin" entry on "payloads" folder?

urherenow, 2016-06-06:
Luma3DS has nothing to do with this. Change/Update your a9lh.

hartmannaf, 2016-06-06:
You could use another bootloader, with screen init.

Also it should not be the problem that ge is closed source, since its a simple a9lh payload.

urherenow, 2016-06-06:
Simple? If you say so. Their loader.dat was only ever partially reversed because it's obfuscated like crazy. Just enough for CFW to be born. Still, many of their methods haven't been reversed, such as their cheat engine and how it hooks pretty much everything that runs on native_firm.Heck even getting different language DLC running didn't happen until recently.

If you can inject screen init to their code, please do, but there's nothing wrong with my answer.

hartmannaf, 2016-06-06:
We are talking about GW's arm9loaderhax.bin, which simply loads their launcher.dat, not from directly loading their launcher.dat. So it doesn't have to do something with closed source or open source, its only a simple arm9loaderhax payload from GW.

billy-acuna, 2016-06-08:
Of course it doesn't have to do something with closed/open source, the payload just loads launchet.dat, but the fact is that launcher.dat (or perhaps GW's payload?) doesn't like no-screeninit or "dark" screeninit, it starts only with full-bright screeninit.That's why only boots with bootctr9 for no-screeninit users.

urherenow, 2016-06-08:
Jesus, you people are dense. Open source matters because you can't inject the screen init code via hex editing.

TuxSH, 2016-06-09:
>  I'm not sure if the Luma config.bin is looked at before chainloading or not
It's looked at [very early](https://github.com/AuroraWright/Luma3DS/blob/master/source/firm.c#L47) indeed.

closed 2016-06-09 by TuxSH

urherenow, 2016-06-09:
Lines 147-150. Chain loading is done right before splash screen stuff is fished out of the config.

AuroraWright, 2016-06-09:
What are you all saying? Luma does screen init before launching a payload... any payload