LumaTeam/Luma3DS #114: LayerFS support

tranaconda, 2016-06-09:
Now that NTR is discontinued and quite frankly, a little painful to deal with (crashes, can't launch FTPd after it was launched, ...), it would be great to have LayerFS support in Luma.

In NTR this meant having a folder with the TitleID of the game in the SD root (though a subfolder would be cleaner) with files in it replacing the ones in RomFS if they exist. This would be a great help for making inofficial translations (I'm on a team to translate Dai Gyakuten Saiban) or other hacks.

TuxSH, 2016-06-09:
Yes of course, I'd be great. It is however difficult.

Don't expect this to happen soon but I'll think about it.

LukeUsher, 2016-06-09:
Until then, there is always the option of rebuilding romfs with the patched files and using HANS, although it is much less user-friendly.

Mori-S, 2016-06-09:
Hans wont run the beyond the labyrinth so I couldn't even use my real card with the translation patch. I ended up having to make a layerFS plugin for it which was a pain in the rear, but thankfully i got that working. Anyway I'm sure whatever you end up putting in luma3ds will end up being easier than the convoluted process I had go through to get it working

Mori-S, 2016-06-09:
Oops accidentally erased my post, but yeah i guess if it would ever run with hans it'd save me the trouble of having to run NTR which I don't like

tranaconda, 2016-06-09:
Yeah, but for quick testing while you're working on it, its more convenient with LayerFS... Thanks for considering it, Tux!

Mori-S, 2016-06-10:
I dunno I just don't like NTR personally, it has nothing to do with the functionality of it really. LayerFS is pretty awesome once it's working

DeathChaos25, 2016-06-21:
This would be amazing tbh, LayeredFS is the only reason I even use NTR, if I could run translations and romhacks for my cartridges without having to deal with a lot of instability due to features I'll never use, it'd be a godsend.

wabberz, 2016-10-08:
I agree this would be amazing, it's very annoying having to reboot all the time when trying out hacks. I realize it's your free time and you have no obligation at all to help us, but this would really help us speed up our translation project. 

In any case, thanks for your hard work on the already amazing Luma3DS!

henke37, 2016-11-06:
I took a look at the source files from the layterfs plugin generator. It appears they simply hook the fsRegArchive and userFsTryOpenFile functions.
[main.c.txt](https://github.com/AuroraWright/Luma3DS/files/573875/main.c.txt)

fox8091, 2016-11-08:
henke37: Where did you get that..?

henke37, 2016-11-08:
https://gbatemp.net/threads/how-to-create-and-use-a-layeredfs-plugin.389977/

wabberz, 2016-11-08:
More precisely, from here: https://github.com/44670/layeredFS/

Margen67, 2016-11-16:
Fixed by https://github.com/AuroraWright/Luma3DS/commit/b5336c81cc82b6c5e8115249342beb5b065cdce9?

henke37, 2016-11-16:
Negatory. That does not implement the "layer" part. The idea of Layerfs is that it doesn't contain the entire romfs. Only the individual files you wish to override.

AuroraWright, 2017-04-14:
Fixed

closed 2017-04-14 by AuroraWright

wabberz, 2017-04-14:
Thank you so much for your hard work! Is this based on Delebile's work?