LumaTeam/Luma3DS #115: Re-enable downgrading from >11.0.0's

Plailect, 2016-06-10:
Either patch the AMPXI:CanInstall or the minimum versions themselves, the end result is the same.

This will make it so downgrading from 11.0.0 no longer requires booting an older firmware (which is a bad crux to rely on since any new update could prevent using older firmware binaries).

[TuxSH's relevant gist](https://gist.github.com/TuxSH/a2ff24d82bdb00ce2c6a35d9cd3f53bc)

ihaveamac, 2016-06-10:
wasn't this already done or am I mistaken?

7ea2a0a278042c919e6ede87091da0f1e608a38f

TuxSH, 2016-06-10:
Yup, this is already implemented (and the 5.5 stable release has this feature).

closed 2016-06-10 by TuxSH

Plailect, 2016-06-10:
Well crap, I managed to miss it looked through the commits and didn't see it in the release notes. Sorry for wasting your time :P

urherenow, 2016-06-10:
Question. Is firm protection disabled when booting 11.0 without a9lh? If not, what good does this do? Wouldn't downgrading with the 11.0 firm staying intact, cause a brick? Or at least prevent svchax from working?