LumaTeam/Luma3DS #117: GBA crash on sysnand 9.2 while OK with EmuNand 11

xuefer, 2016-06-11:
the setup is a9lh. with SysNand == 9.2 and EmuNand == 11
the game will crash if i install it to EmuNand only. I've learn that i have to install the GBA cia into both nand so i did and it works. but when i start it from SysNand 9.2 it still crash. could it be that 9.2 is not supported for gba patch?

strange is that with Luma3DS+EmuNand 11, i can launch my DS-TTi NDS flash cart. it won't run with 8.5 or 9.2 system which leads to crash screen

urherenow, 2016-06-11:
I bet your twl and/or agb firm is/are patched on your sysnand, and your emunand has the stock firmware since you upgraded it. Install an unpatched twl_firm and agb_firm to sysnand, and you'll be fine.

AuroraWright, 2016-06-11:
https://github.com/AuroraWright/Luma3DS/wiki/Other-features-and-notes#standard-features

closed 2016-06-11 by AuroraWright

xuefer, 2016-06-11:
i never patch it, maybe the sell have it patched already, but it's me that do a9lh installation.

I did read the clean firm to use NANDFIRMs.7z and i re-check by installing them using 2.1 version of FBI as the wiki said. i also extract files from 9.2.0-20J(Full)_n3DS.zip (from https://github.com/Plailect/Guide/wiki/Part-2-(9.2.0-Downgrade)) and run PlaiSysUpdater_v04.zip to apply them, i get "DOWNGRADE COMPLETE", yet gba still crash

TuxSH, 2016-06-12:
> FIRM reboots replacement **(if you're not booting a 9.0/9.2 NAND)**, which allows these to work:
3DS games which work in the larger RAM mode (Smash Bros, Monster Hunter 4... on Old 3DS)
Custom GBA games
Old DS flashcards

xuefer, 2016-06-12:
exactly, thanks for outlining it. I was scanning the document too fast and worry more about installing original firm