LumaTeam/Luma3DS #124: possible unrelated?

Patacakes, 2016-06-18:
I had an issue the other day.  I bought SMB3 on eshop for N3DS VC.   tried the download play option with my friend. i had EU console they have US.  download play would not detect the game. 
I thought that maybe the game was region locked to share with only the same region consoles. 

so i installed the CIA for the US version.  but now when i start the game, and press Start Download play.  it tells me that the wifi is disabled. even though it is not. 

is this a bug? or a sly game patch checking that the console is the same region as the VC game installed?

(I`m on N3DS EU latest Luma build. (not nightly/ or Dev)

Subv, 2016-06-19:
I think the DLPChild NCCHs are region locked, but i might be wrong.

fox8091, 2016-06-23:
Subv: You're not wrong. DLP is region locked too.

closed 2017-03-06 by TuxSH