LumaTeam/Luma3DS #125: Compiling Luma3DS with latest ctrulib from smea (commit: 1cef3e5) it errors out

PeggerWed, 2016-06-18:
When compiling your source with latest ctrulib (commit 1cef3e5) from smealum you get the following errors:

(Note, this is for everything that uses CakeBrah/Brahmaloader (libkhax))

`khaxinit.cpp
arm-none-eabi-g++ -MMD -MP -MF /d/ABuildCFWs/Luma3DS/CakeBrah/build/khaxinit.d -
g -Wall -Wextra -O3 -mword-relocations -fomit-frame-pointer -ffast-math -march=a
rmv6k -mtune=mpcore -mfloat-abi=hard -I/d/ABuildCFWs/Luma3DS/CakeBrah/include -I
/c/devkitPro/libctru/include -I/d/ABuildCFWs/Luma3DS/CakeBrah/build -DARM11 -D_3
DS -DARM_ARCH -DLAUNCHER_PATH='"Luma3DS.dat"' -fno-rtti -fno-exceptions -std=gnu
++11 -c /d/ABuildCFWs/Luma3DS/CakeBrah/source/libkhax/khaxinit.cpp -o khaxinit.o

d:/ABuildCFWs/Luma3DS/CakeBrah/source/libkhax/khaxinit.cpp: In function 'Result
KHAX::IsNew3DS(bool*, u32)':
d:/ABuildCFWs/Luma3DS/CakeBrah/source/libkhax/khaxinit.cpp:985:47: error: cannot
 convert 'u8* {aka unsigned char*}' to 'bool*' for argument '1' to 'Result APT_C
heckNew3DS(bool*)'
   if (Result error = APT_CheckNew3DS(&isNew3DS))
                                               ^
make[2]: *** [khaxinit.o] Error 1
make[1]: *** [build] Error 2
make: *** [out/3ds/Luma3DS] Error 2`

TuxSH, 2016-06-18:
Change: ```c
u8 isNew3DS = 0;```
to:
```c
bool isNew3DS = 0;```

I'm waiting for @Myriachan to fix this.

PeggerWed, 2016-06-18:
Ok, thanks, this does indeed work for Luma3DS and CakesFW and other CFW's

But if gives another warning on all (but it compiles)

```
arm-none-eabi-gcc -MMD -MP -MF /d/Z_Testing/Luma3DS/CakeBrah/build/brahma.d -g -
Wall -Wextra -O3 -mword-relocations -fomit-frame-pointer -ffast-math -march=armv
6k -mtune=mpcore -mfloat-abi=hard -I/d/Z_Testing/Luma3DS/CakeBrah/include -I/c/d
evkitPro/libctru/include -I/d/Z_Testing/Luma3DS/CakeBrah/build -DARM11 -D_3DS -D
ARM_ARCH -DLAUNCHER_PATH='"Luma3DS.dat"' -c /d/Z_Testing/Luma3DS/CakeBrah/source
/brahma.c -o brahma.o
d:/Z_Testing/Luma3DS/CakeBrah/source/brahma.c: In function 'get_exploit_data':
d:/Z_Testing/Luma3DS/CakeBrah/source/brahma.c:61:18: warning: passing argument 1
 of 'APT_CheckNew3DS' from incompatible pointer type [-Wincompatible-pointer-typ
es]
  APT_CheckNew3DS(&isN3DS);
                  ^
In file included from c:/devkitPro/libctru/include/3ds.h:31:0,
                 from d:/Z_Testing/Luma3DS/CakeBrah/source/brahma.c:1:
c:/devkitPro/libctru/include/3ds/services/apt.h:464:8: note: expected '_Bool *'
but argument is of type 'u8 * {aka unsigned char *}'
 Result APT_CheckNew3DS(bool* out);
```

TuxSH, 2016-06-19:
same thing here

fox8091, 2016-06-20:
Tux: Change it to 
`bool isNew3DS = false;`
instead. Also, change
`*answer = isNew3DS != 0;`
to
`*answer = isNew3DS;`

fox8091, 2016-06-20:
I made a PR to libkhax a little while back.

PeggerWed, 2016-07-01:
Still same problem :/
D0k3 has updated all his tools and they all compile fine under latest ctrulib.

fox8091, 2016-07-01:
The submodule still hasn't been updated..

Myriachan, 2016-07-02:
Did I merge the commit that you wanted to do to libkhax properly?  Let me
know if there is something that you need.

PeggerWed, 2016-07-02:
Ok, latest commit compiles without errors on latest ctrulib, except for the 1 warning (see previous posts)
_(Latest commit 9621181  @TuxSH TuxSH Use bool instead of u32 where it's relevant)_

TuxSH, 2016-07-02:
Fixed.

closed 2016-07-02 by TuxSH