LumaTeam/Luma3DS #128: DS and DSi ware not working

iamapie13, 2016-06-21:
Hello, I decided to instal the arm9loaderhax versión of luma and now my DSi and DS games show black screen, I tried to instal a TWL_FIRM patch, but it didn't work

ihaveamac, 2016-06-21:
if "arm9loaderhax.bin" is not Luma3DS (boot manager, etc), you need to use the pathchanger.
https://github.com/AuroraWright/Luma3DS/wiki/Installation-and-Upgrade#optional-making-luma3ds-work-with-an-external-boot-manager

installing a TWL_FIRM patch doesn't help, instead you should undo that.
https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-Troubleshooting#i-have-issues-with-gbadsdsiware-games-and-i-have-installed-a-patched-agbtwl-firm-cia-in-the-past

TuxSH, 2016-06-21:
Please read the wiki before asking (@ihaveamac even made that easier for you)

closed 2016-06-21 by TuxSH

iamapie13, 2016-06-21:
Thank you for the help, I'll try to undo the TWF patch, I only need the ID of the CIA to get it from Nus, but I can try to figure that on my own

ihaveamac, 2016-06-21:
if you actually read the second wiki page I linked you to, it gives you a link to TWL_FIRM.........

iamapie13, 2016-06-21:
sorry, my apologies, I forgot to see it, I was rushing in school