LumaTeam/Luma3DS #129: just making sure

monkeyman4412, 2016-06-21:
does this work on 11.0.0-33u without any problems or you know bricks, (I fully understand there is always a small chance for a brick but I mean is there a less chance)

cheatfreak47, 2016-06-21:
Where on this github does this section say "Help and Support"?
Do not ask questions in the issues section, this is for reporting bugs and suggesting features.

Luma3DS can run EmuNANDs running 11.0.0-33u and can also run 11.0.0-33u on SysNANDs via Arm9LoaderHax.

You can not hack 3DS consoles on stock SysNAND 11.0 without a hardmod beyond userland homebrew. 

Now close this issue, if you have farther questions go to GBATemp or the 3DShacks Subreddit. 
This is NOT the place for it.

JeanpiASDF, 2016-06-21:
Honestly, this stuff with people posting stuff like this on the Issues section is getting out of hand.

closed 2016-06-21 by RShadowhand

TuxSH, 2016-06-21:
Indeed