LumaTeam/Luma3DS #132: Battery Life Depleted in Sleep Mode when overclocking N3DS

wolfpacleader1986, 2016-06-26:
I have my N3DS set to "Clock+L2" and I noticed that since then, my battery depletes very quickly, only getting about four hours on a charge, including sleep mode. So, my guess is that there needs to be something written in the code, a fix that corrects the Clock+L2 so that it turns off and goes back to normal in Sleep Mode.

Shadow322, 2016-06-26:
No just use NTR CFW and turn it off and on when needed

TuxSH, 2016-06-26:
What do you expect, you're raising the power consumption of your N3DS...

Just turn this option on only when needed.

closed 2016-06-26 by TuxSH

cheatfreak47, 2016-06-26:
I know this was closed, but I never actually experienced this sort of battery drain, I've never noticed even a slight difference actually.