LumaTeam/Luma3DS #133: FAQ has viruses

timbiotic, 2016-06-26:
I cannot fix my O3DS because the fix files in the FAQ all have Trojan viruses.  Can they be reposted somewhere safe like mega?

"1.Get the NTR FIRMs package from here and the NAND FIRMs package from here "

piratesephiroth, 2016-06-26:
do you really think your antivirus has the amazing capability of detecting 3DS malware

timbiotic, 2016-06-26:
thanks that is helpful, I have tried it on many and each one lists Trojans. I then downloaded it on my ipad and it says ifile says the archive is invalid. I think it is zippyshare as it also sent me to malware pages.

tranaconda, 2016-06-27:
https://www.virustotal.com/de/file/f74c99eafc59e728e6b8a15cfbb3bed51c103742aafe76c9690f93c0e7a0bf63/analysis/
https://www.virustotal.com/de/file/cbc55eb286c5603e241f14efa1e203a8f98b0716964e660a23bf11dd38949077/analysis/

They are both clean as a whistle, @piratesephiroth was hinting at that already. You have to click the BIG ORANGE button on the TOP RIGHT. Other buttons must be some kind of ad. You should probably install an adblocker.

timbiotic, 2016-06-27:
Windows defender blocks every time.  I guess I'll install Linux. That is clicking the large orange square in upper right corner, right? Can someone email me the files or post on mega?

tranaconda, 2016-06-27:
Use Chrome or Firefox and a decent adblocker like µblock origin

timbiotic, 2016-06-27:
I used my linux box with wget but windows 2012 r2 and windows 10 hates it

closed 2016-06-27 by timbiotic