LumaTeam/Luma3DS #137: Keybinding - bind Decrypt9 to "A" on boot

JHGitty, 2016-06-28:
I want to bind the key "A" to Decrypt9WIP.bin
but I cannot find out how to do it. Rename to `a_Decrypt9.bin` does not work.
*But if I name it `x_Decrypt9.bin` it works fine on key `X`!* How to do it for key `A` ?

ihaveamac, 2016-06-28:
hold L+A

closed 2016-06-28 by TuxSH

JHGitty, 2016-06-28:
But how can I set it only to "A"? Is there any official documentation about keybinding for Luma3DS or do we need to create this kind of documentation?

ihaveamac, 2016-06-28:
you can't since Luma3DS uses A normally. applies to select and R too (maybe B, can't remember)

JHGitty, 2016-06-28:
Okay, what does Luma3DS do with the A key? Can I do something with it? I think there should be any good documentation - no documentation about this yet, right?

JHGitty, 2016-06-28:
Thanks. I have found the documentation part: https://github.com/AuroraWright/Luma3DS/wiki/Other-features-and-notes#payload-chainloading