LumaTeam/Luma3DS #138: GDB stub

henke37, 2016-06-29:
Implement the GDB stub protocol. This would allow for really easy debugging of homebrew and retail titles alike.

NTR already has something similar, but using a nonstandard protocol that nearly no tools understand. GDB on the other hand is well understood by not only gdb itself, but also powerful tools such as IDA Pro.

Oh and "debugging" also means cheating by memory peeks and pokes.

mariogamer2, 2016-07-01:
GDB? What is this (sorry I do not follow NTR) ?

henke37, 2016-07-01:
Yup! That's gdb. It's the debugger that goes with gcc. And gcc is the main compiler for homebrew. As such, it would be very handy to have a gdb stub. It would let you debug homebrew on the console over a TCP connection.

dragos240, 2016-11-17:
No idea if issues get bumped like threads do but this is a fantastic idea and would be incredibly useful for a homebrew developer.

Mrrraou, 2016-11-18:
We are developers too. We understand how useful it is.
However, there are limitations on the consoles (especially O3DS consoles) and other things that are preventing this to be developed for now.

henke37, 2016-11-18:
And what would these limitations be exactly?

TuxSH, 2017-06-05:
(∩ ͡° ͜ʖ ͡°)⊃━☆ﾟ

closed 2017-06-05 by TuxSH

henke37, 2017-06-05:
In conclusion, were there any limitations, then the recent boot9strap attack took care of them. Gdb is in!