LumaTeam/Luma3DS #140: System crashed when start MHX after quit a custom cia

ITJesse, 2016-07-02:
When I run a custom cia like FTP-GMX or BootNTR, and then I can't run MHX.
System will crashed.

But if I run MHX directly, everything is good.

[image]

[crash_dump_00000002.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/344669/crash_dump_00000002.dmp.zip)

ITJesse, 2016-07-02:
I have update to the latest commit (2d6debddb92aacb0f54e5a1701c5f8866dbaac29), and have no help with this problem.

TuxSH, 2016-07-02:
Did you try closing the homebrew CIA app before launching MHX (just curious).

hyln9, 2016-07-02:
@TuxSH I encountered a similar problem on a New JPN 3DS, and had no luck booting anymore.

ITJesse, 2016-07-03:
@TuxSH not the homebrew CIA.
The homebrew will display some color screen when they start.
But FTP-GMX isn't.

TuxSH, 2016-07-03:
Are you getting the same problem with other homebrew CIAs?

TuxSH, 2016-07-03:
Do you have the same problem with the latest commit on `master` branch?

ITJesse, 2016-07-03:
EasyRPG Player

onmyouji, 2016-07-03:
Isn't this a common problem with FTP-GMX?

If you load CIA games or homebrew launcher after exiting FTP-GMX, strange thing will happen. I also had problem when loading Fire Emblem after exiting FTP-GMX.

Check this [GBAtemp post](https://gbatemp.net/threads/release-ftp-gmx-customize-your-ftbrony.410506/page-4#post-6283207) and other posts after that. A user actually found out that the problem is related with libsf2d component of FTP-GMX.

closed 2016-09-04 by AuroraWright