LumaTeam/Luma3DS #147: [Question] What's this title?

v0idst4r, 2016-07-24:
Hello. After installing Luma3DS this strange title appeared on my home menu. I'm not sure if it is a title that Luma3DS installed for region free, etc but I was curious because it doesnt show on any of my other CFWs. http://i.imgur.com/nJIp52W.jpg

ihaveamac, 2016-07-24:
check with FBI 2.x to see if anything unusual was installed to NAND? maybe like something with a title ID that begins with 00040000?

v0idst4r, 2016-07-24:
A wild 0x0004000000686200 with the product code CTR-HB-LUMA has appeared

v0idst4r, 2016-07-24:
Seems to be some kind of thing for Luma. Maybe RF as I suggested

ihaveamac, 2016-07-24:
that's probably not supposed to be installed to NAND.

v0idst4r, 2016-07-24:
Actually that doesn't seem to match the title shown in the image.

ihaveamac, 2016-07-24:
things installed to NAND that shouldn't will usually show a black icon with no name. "000400000068620"/"CTR-HB-LUMA" doesn't look like something that should be in NAND so might want to delete it.

v0idst4r, 2016-07-24:
Actually I misread your comment and was reading that from SD. I don't see anything out of the ordinary in NAND

closed 2016-08-03 by TuxSH