LumaTeam/Luma3DS #148: Cannot chain payloads

v0idst4r, 2016-07-25:
Hello! I recently did the long process of dumping my OTP.bin for A9LH and installed Luma as my main payload. I was curious how to chain payloads and someone in #3dshacks told me to add the payload to /luma/payloads/KEY_NAME.bin. I also read this up on the wiki. I'm trying to make start go to Decrypt9 but the payload does not load at all and just loads the CFW. I'm not sure if I misspelt something or start is already used etc. And if it is already used it should be noted on the wiki. Here's the folder structure http://m.imgur.com/D9ZSOKx

v0idst4r, 2016-07-25:
Oh I also have tried renaming it start.bin

ihaveamac, 2016-07-25:
have you tried something like `start_Decrypt9WIP.bin`? so far it seems you've tried `start_Decrypt9WIP.dat` (.dat is incorrect) and `start.bin` (doesn't have a name).

v0idst4r, 2016-07-25:
Yeah I realized I accidentally renamed it to dat. I've tried it with the bin in the Decrypt9 release too.

ihaveamac, 2016-07-25:
tried another button? I use `left_d9.bin` holding left just fine.

v0idst4r, 2016-07-25:
I just tried with "a" still no dice. Would a video help?

ihaveamac, 2016-07-25:
are you using `Decrypt9WIP.bin` and not the dat? are you adding a name to it (just "`a.bin`" won't work I think)?

v0idst4r, 2016-07-25:
Yes I am using Decrypt9.bin

v0idst4r, 2016-07-25:
The reason I need to do this is because I can't load Decrypt9.3dsx in Luma cfw, it just goes back to the hbl

v0idst4r, 2016-07-25:
I've tried many more attempts than the ones in this thread. I've tried a_D9.bin a.bin start_Decrypt9.bin

pbanj, 2016-07-25:
Try this
https://cdn.discordapp.com/attachments/205039685859278848/207038532676485121/start_decrypt9WIP.bin

ihaveamac, 2016-07-25:
if you're using "A" you need to hold L+A to use the payload. start should work. try another button as well, like left.

v0idst4r, 2016-07-25:
http://m.imgur.com/9ZXTHBw Strange the file on the left didn't work but the one you sent worked fine. Does it not like NAME to start with a capital?

pbanj, 2016-07-25:
I think you may have been using the DAT file and not the bin

pbanj, 2016-07-25:
Or your bin was corrupted somehow

v0idst4r, 2016-07-25:
As you can see they are the same size. The dat is 223kb cmon guys a simple extension rename mistake...

v0idst4r, 2016-07-25:
Yeah might've been corrupted. A lot of my files that I put on my SD have randomly been becoming 0bytes

pbanj, 2016-07-25:
Try giving the one I linked a capital D and see if it still loads, if not we known that was it

v0idst4r, 2016-07-25:
Currently in the middle of a nand backup :(

v0idst4r, 2016-07-25:
I'll update when its done

pbanj, 2016-07-25:
So did the capital letter work also?

v0idst4r, 2016-07-25:
Nope. The caps is most likely what's causing it. I wrote another payload like corbenik and I had to not use caps

v0idst4r, 2016-07-25:
https://github.com/AuroraWright/Luma3DS/blob/af19719e3463dc4adf16438e6473b9401017ea0f/source/fs.c

I haven't done a lot of file system oriented things in C so I don't seem to be able to find where it is ignoring caps...

v0idst4r, 2016-07-25:
For example a_decrypt9WIP.bin will work but not a_Decrypt9.bin

v0idst4r, 2016-07-25:
Actually after testing it worked half the time wtf?

v0idst4r, 2016-07-25:
Weird sometimes the payloads doesn't work but it must not be caps.

closed 2016-07-25 by v0idst4r