LumaTeam/Luma3DS #149: Can't load homebrew like Decrypt9?

v0idst4r, 2016-07-25:
Hello I'm trying to run Decrypt9 through hblauncher_loader but Decrypt9 just black screens. I've also tried loading the dat in ctrbootmanager but that froze also. Homebrew like FTPd works but nothing like EmuNAND9 or Decrypt9. I need a way to install a CIA manager to my EmuNAND and I was trying to do it with H&S inject. Is there a possibility it could be a patch in the CFW preventing it from loading and will there ever be a mode to load the CFW without any patches at all? The way I'm trying to load it through BrowserHax on 9.1JPN.

v0idst4r, 2016-07-25:
Oh also, what's weird is I used FBI through BrowserHax to install hblauncher_loader

Ev1lbl0w, 2016-07-29:
Decrypt9 only works on 9.2 and below, I think. So, if you don't have A9LH to run it as a payload, I think you must downgrade, enev with CFW.

v0idst4r, 2016-07-29:
No the sysnand is on 9.1. And the sysnand is what the cfw is running.

v0idst4r, 2016-07-29:
You didn't read the OP did you?

hartmannaf, 2016-07-31:
You could try to use the latest version of decrypt9 WIP.

SerLizar, 2016-08-06:
from Plailect's guide "Copy Decrypt9WIP.bin to the /luma/payloads/ folder on your SD card and rename Decrypt9WIP.bin to x_Decrypt9WIP.bin" then you can follow the instruccions on [part 4 section II of that guide](https://github.com/Plailect/Guide/wiki/Part-4-(2.1.0-Downgrade)#section-ii---injecting-fbi) you just have to press x while booting your 3ds to boot directly into Decrypt9.

EmuAGR, 2016-08-06:
You'll need FW <9.2 AND signature patches enabled on your SysNAND for D9 to work. Your current CFW launches on SysNAND, but patches EmuNAND. You need a SysNAND CFW like MiniPasta to disable sigchecks on <9.2 SysNAND.

v0idst4r, 2016-08-06:
I seriously think none if you can read. I said I'm on 9.1. Since this will never be resolved due to a lack of brain cells I'll just close it.

closed 2016-08-06 by v0idst4r

camatthew88, 2017-02-12:
Its okay

camatthew88, 2017-02-12:
Try using slowhax or savehax

Margen67, 2017-02-16:
*https://3ds.guide/

v0idst4r, 2017-02-21:
@camatthew88 I've already solved it, thanks though.

urherenow, 2017-02-21:
Bump? Really? This isn't a forum. There's no such thing as a bump, and this issue was closed 7 months ago (ironic, since every reply afterwords suggests things that didn't even exist when this issue was closed)

SMH

v0idst4r, 2017-02-21:
xd