LumaTeam/Luma3DS #150: Manual of Out of Region Games do not work

ederenzi78, 2016-07-26:
I have a EUR O3DS with A9LH and Luma 5.5 on SysNand. When I try to load the Manual of USA games, an error message pops up. Game has been downloaded using CIAangel and I created an entry in the locales directory to correctly handle the game. I tried for example "Sega 3D Classics collection" or "Kokuga".

TuxSH, 2016-08-03:
Maybe it's just because the corresponding files from your region/language are missing?

ederenzi78, 2016-08-03:
But I created a file in the locales directory to specify Region and Language for the title. In this case USA EN for a USA game. Luma 3ds correctly sees this file (infact, without it, Sega 3d classic does not save anything) but probably this is not seen for the manual .... I don't think CIAngel does not download the manual as it should download everything related to the title ...

hartmannaf, 2016-08-04:
From what I know the current region free menu patch is the problem, because it only patches the home menu to allow applications from other regions, but it is not skiping the check for manuals.
So to be able to use out of region manuals we would need a addition or new patch for the home menu.
This was also a problem when using it with the FreeMultiPatcher.

TuxSH, 2016-08-04:
Are there really checks for manuals though?

piratesephiroth, 2016-08-05:
Manuals don't have icons therefore they're always region-free.

hartmannaf, 2016-08-05:
Its possible to upen manuals ingame, but homemenu will give you an error, if its an out of region game. If you use NTR, is possible to open all menuals, without language emulation from what I know, so there needs to be a check somewhere.

ederenzi78, 2016-08-06:
Can confirm that current region free menu patch is the problem, as with NTR manuals are correctly opened

Margen67, 2016-08-16:
Why not just look through the source co- oh wait..

d0k3, 2016-12-07:
Okay, my 5 cents. Manuals don't have a region that can be patched. The region is inside the icon (nowhere else) and CFAs like the manual have no icon. The manual also strangely works when there is a 'hard' regionfree patch inside the CXI icon (this means an edit of the actual CXI file, which is done by several CIA creator tools, f.e.). 

The issue is there for all out of region installs that don't have that regionfree patch, which includes, f.e. freeShop installs and CIAs created via several tools (like D9).

sifiro, 2016-12-07:
But it's a little strange... (this is related with previous Issue) I Checked that my Cart (USA in EUR Console) can run manual out-of-region without any modification in cart. Musn't works the same patch for carts but for CIAs? if is a icon CXI Issue (and should have same or similar structure, It is not like this?).

hartmannaf, 2016-12-07:
Which games was it, maybe the game is regionfree(like the pokemon games).

sifiro, 2016-12-07:
Senran Kagura 2: Deep Crimson (USA) and... I'm Pretty sure is Region Lock

AuroraWright, 2017-06-28:
Should be fixed

Hikari-chin, 2017-06-28:
To add onto what @AuroraWright said, manuals are now properly region patched as of commits 9f78d7f and b88dc9a.

closed 2017-06-28 by Hikari-chin