LumaTeam/Luma3DS #151: Patch ssl:C to disable cert verification

ChlorideCull, 2016-07-28:
Useful for working with data mining applications. I assume patching `CreateContext` to always apply bit 9 ("Disables server cert verification when set." according to 3dbrew) would do the trick?

Should obviously come with huge warning signs for the configuration option, since it would create a security nightmare on untrusted networks, like public WiFi.

PokeAcer549, 2016-07-31:
Perhaps it'd have to be enabled manually, so you cant accidentally do it :stuck_out_tongue:

thejsa, 2017-04-19:
Bump as this would be absolutely amazing for reverse engineering and the like

Margen67, 2017-05-01:
https://github.com/SciresM/3DS-SSL-Patch ?

PokeAcer549, 2017-05-01:
Literally only came out 14 minutes ago, but yes that's fine ~~except you have to build an old copy of Luma~~ - this can be closed then.

closed 2017-06-03 by TuxSH