LumaTeam/Luma3DS #153: Can't Play DSiWare Games

Thewanildo, 2016-08-06:
I don't use any kind of bootloader and I still can't play DSiWare games.

astronautlevel2, 2016-08-10:
Can you provide more details?

Are you on the latest stable? If not, which commit are you on?

Are you on the regular or developer branch?

Thewanildo, 2016-08-15:
I'm on the latest stable version and I also have a9lh installed.
I can play any Virtual Console game normally and any 3ds game without a single problem.

Mrrraou, 2016-08-15:
Are you using the arm9loaderhax.bin ? Is the path patched, if it's not on the root on the SD card and called "arm9loaderhax.bin" ?

MelonGx, 2016-08-16:
Have you read the whole wiki even the troubleshooting page?

You should know Luma conflicts TWL/AGB patches.
It is highly recommended that TWL/AGB patches should be removed if you use Luma.

Thewanildo, 2016-08-16:
I don't use TWL/AGB patches.
PS: I don't know what I did but instead of getting the error message now I get a black screen.

MelonGx, 2016-08-17:
Then some simple questions:
1) SysNAND or EmuNAND? TWL/AGB is not supported for unlinked EmuNAND.
2) Same Region? It may be the problem if you play out-of-region TWL/AGB.

urherenow, 2016-08-17:
Thought TWL region patches were taken care of ages ago. You sure about that @MelonGx?

Thewanildo, 2016-08-17:
Unlinked RedNand and I thought that Luma was Region Free.

Margen67, 2016-08-18:
Why are you using Rednand?

Thewanildo, 2016-08-18:
Because the tutorial I followed told me to :v

Margen67, 2016-08-18:
If you already have A9LH you should switch to updated SysNAND

AuroraWright, 2016-09-04:
You can't play DSiWare from emuNAND unless you have them on both NANDs

closed 2016-09-04 by AuroraWright