LumaTeam/Luma3DS #155: luma 3ds won't show open button for games/apps?

sketchy1, 2016-08-11:
Can someone help me with luma? 
My emunand crashed saying the sdcard has been removed, and when I booted emunand again the open button at the bottom of the screen was gone, so now I cant open any games in emunand running 9.2 sysnand 10.1 emunand . also cant press the a button to open anything and double tapping an app in the touch screen would cause it to freeze

james-d-elliott, 2016-08-11:
Can attempt moving all of the folders out of the Nintendo DS folder on your SD card, this has fixed similar issues for me before.

dogtopus, 2016-08-18:
roxas75/rxTools#475
Well, I don't think your problem was CFW-related. You probably need to ask someone else for help.

Margen67, 2016-08-18:
>emunand
Well there's your problem

closed 2016-08-28 by TuxSH