LumaTeam/Luma3DS #161: Enable game update when 3ds is on 10.7-11.0 with non-official soft.

mariogamer2, 2016-08-18:
Lot of user report (me too),that,when having non-copyrighted software on the 3ds (FBI,JKSM,SysUpdater),game update are somewhat "disabled".

Does it is possible to enable hem?

Mrrraou, 2016-08-19:
non-copyrighted software != unsigned software
And what do you mean by that ? There are no software updates on the eShop for these.

james-d-elliott, 2016-08-20:
You can game update via the eShop. The issue is that you cannot update USA titles on a EUR console because the DS downloads the EUR region update and the USA region game doesn't look for those files. When you launch it should ask you, if you have the EUR title and EUR console, you will update fine.

einstein95, 2016-08-28:
This is because afaik the game updates using the MINT module which has its own region-locking due to signing into Nintendo's services, basically acting as a mini eShop.

closed 2016-08-29 by TuxSH