LumaTeam/Luma3DS #162: ARM9 Exception when Booting a9lh normally.

jdcox215, 2016-08-20:
Had a9lh installed from the guide before Red Nand was involved. When I installed, the guide instructed me to create a folder named S when on Sysnand, making sure not to update it when you saw the S. Anyways followed the instructions to the T, and it was working great.
Now newer version of A9LH is out, and I went to update, both A9LH and to Luma 6 dev.
It's giving me an ARM9 exception when trying to start up normally.
Am attaching the contents of last dump file.
[image]

TuxSH, 2016-08-20:
Weird. To be sure, which download link did you use?

jdcox215, 2016-08-20:
The latest dev. Switching back to the regular version is letting me boot! But I'm wondering, also... I installed the Pokemon Bank CIA and it said it had an update, so I went to update it, and it gave me an error code. (Which I will try and update again and tell you the error code) but it was really weird

TuxSH, 2016-08-20:
Hmm. By latest dev, you mean latest release, or...?

jdcox215, 2016-08-20:
The one, that when you click on releases, it says Luma3DSv6.0.1-dev.7z

TuxSH, 2016-08-20:
Okay, thanks

TuxSH, 2016-08-20:
Are you sure you used that file?

jdcox215, 2016-08-20:
100% positive

TuxSH, 2016-08-20:
Hm, really? Please try this one out.
[arm9loaderhax.zip](https://github.com/AuroraWright/Luma3DS/files/428446/arm9loaderhax.zip)

jdcox215, 2016-08-20:
That's ok. I don't need the dev version of Luma any ways

jdcox215, 2016-08-20:
I got it working but whenever I try and install an app on free shop the app quits..

TuxSH, 2016-08-20:
Did you try what I sent you though (even if you don't need it, I'd like you to try it and see if it crashes)

fox8091, 2016-08-20:
Him saying to try that version wasn't for you specifically, but to diagnose the issue for others that may be having it so it can be fixed in source. Please try it

jdcox215, 2016-08-20:
Ok So i'm gonna try the 6.01 dev version with the arm9loaderhax.bin file you sent to me in the chat and I'll let you know what happens

jdcox215, 2016-08-20:
OK so I used the Luma3ds.dat in the 6.0.1 zip file, and the ARM9loaderhix bin that you linked to in the comments, and that produced the same result. The exception

TuxSH, 2016-08-20:
Please send a picture of it

jdcox215, 2016-08-20:
[image]

TuxSH, 2016-08-20:
Could you send me the produced dump file, please ?

jdcox215, 2016-08-20:
Yeah it will be a lot later. I will have to produce the error again because I switched back to the one that works so I could play a game.  

I will reproduce the error and send you the dump file when I get off work. Around 10 eastern time

TuxSH, 2016-08-20:
okay, ty :D (by any chance, was there a firmware.bin inside `/luma/`)?

jdcox215, 2016-08-20:
Yeah was I supposed to delete when switching from normal back to dev version??

TuxSH, 2016-08-20:
The thing is, that master will only use firmware.bin in a few cases, while the dev branch will always use firmware.bin. It's possible that this firmware.bin is corrupted.

jdcox215, 2016-08-20:
Ok! I will try that when I get home!! Thanks for the help

jdcox215, 2016-08-21:
Alright, so I don't get it...... I did a fresh update of A9LH, just in case something got corrupted there.... Then, I deleted the firmware.bin in the luma folder, and the config.bin also, (because I knew I was going to have to reconfigure it).   I then copied the luma3ds.dat from the dev folder, and then used the arm9loaderhax.bin that you gave me in the messages. Booted it, and it worked!!! :)

fox8091, 2016-08-21:
It was likely just the firmware.bin

jdcox215, 2016-08-21:
Ok so I was able to boot normally with the dev release, but now, unfortunately I cannot install any CIAs with FBI.  

I tried it only with a Kirby game, but it would load all the way then give me an exception, which, when I get home from church, I will reproduce and send you the dump file. Also where would you like me to send you the dump file?? I can't attach it on the messages

Mrrraou, 2016-08-21:
https://transfer.sh/

jdcox215, 2016-08-21:
Here's my crash dump I got
https://transfer.sh/t4848/crash-dump-00000000.dmp

closed 2016-09-11 by AuroraWright