LumaTeam/Luma3DS #164: NTR 3.4 doesn't work on o3DS 11.0 firmware

TarkinMX, 2016-08-21:
NTR 3.4 on o3DS doesn't work on 6.0.1, I had to revert back to 5.5 to get functionality again. Also, I've been having to use firmware.bin just to get it to work on 5.5 so I don't know if it's a problem with NTR or Luma3DS

Shadow322, 2016-08-21:
Update your bootntr and it will work on 6.0.1 without firmware.bin

closed 2016-08-21 by TuxSH

TarkinMX, 2016-08-22:
Apologies, I meant to say NTR 3.2 and I can't update any higher, I'm on and Old 3DS XL. As can be seen on https://github.com/44670/BootNTR/releases 3.4 preview 1 and 2 clearly state "This is a New3DS-Only preview release. Do not try it with Old3DS."

Shadow322, 2016-08-22:
Im not talking about the ntr.bin im talking about the BootNTR cia and yes u can use NTR 3.4 preview 2 on O3DS