LumaTeam/Luma3DS #166: Latest non-dev build won't boot 11.0.0-33E without firmware.bin

lyptt, 2016-08-23:
I followed plailect's guide to get my (previously downgraded) 4.5.0.8E old 3DS onto Luma3DS CFW with arm9loaderhax, and I've reached the point where everything's finally working, with one exception.

Unless I put the firmware.bin for my old 3DS in the /luma folder, Luma3DS considers my firmware to be too old and tells me to use a firmware.bin file.

I'm 100% for sure on the latest 3DS firmware, and my arm9loaderhax works perfectly fine. Any ideas what I'm doing wrong?

TuxSH, 2016-08-23:
Downgraded with Gateway? If so, follow: https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-Troubleshooting#i-get-a-black-screen-on-boot-with-an-old-3ds-which-has-been-downgraded-from-the-gateway-menu-in-the-past

lyptt, 2016-08-23:
Yeah I think I might've? I'm not 100% sure, but I know I performed a downgrade to 4.5.0 early this year before all the CFW stuff came out. That was back when you had to run a browser exploit every time you wanted emuNAND.

lyptt, 2016-08-23:
Actually I did remember having to do this anyway to stop Luma getting a black screen on my redNAND. I should've remembered really... I'll give this a try, thanks!

lyptt, 2016-08-23:
Yay that worked! Thanks for the help on this. That gateway downgrade's been a huge pain in my ass this entire time. So much regret :/

closed 2016-08-23 by lyptt