LumaTeam/Luma3DS #170: TwlBg broke the DSTWO card

ITJesse, 2016-08-26:
In version 6.0.1, if I place the TwlBg.cxi in the luma folder, the DSTWO cannot start any game.

In version 6.1, I can't use it anymore.

Maybe you should provide a switcher for this feature.

Shadow322, 2016-08-26:
No your shits broken because it works here

ITJesse, 2016-08-26:
It freeze in a black screen when I select a game.

And everything gose well when I switch to 6.0.1 without TwlBg.cxi.

ThunderRemix, 2016-08-26:
Although unrelated to the issue in a way, Pokemon Heart Gold freezes on a black screen after selecting and loading the save.

cheatfreak47, 2016-08-26:
Seems to work fine on my end. I have a late model DSTwo, the one that was actually a DSTwo Plus, but has the old DSTwo Label, with no screw on the back, it's held together with glue I think.

@ThunderRemix Your save file is probably the problem, Heartgold is working fine for me.

takeshineale128, 2016-08-26:
Also white screen on Professor Layton and the Lost Future retail card

cheatfreak47, 2016-08-26:
@takeshineale128 I can confirm that one, happens for the DSTwo as well for that game, switc

ThunderRemix, 2016-08-26:
@cheatfreak47 Then why does it work without TwlBg patches?

dankzegriefer, 2016-08-26:
This is not the correct place to report this. TwlBg is not an official part of Luma.

Margen67, 2016-08-26:
@dankzegriefer The patches were added to Luma.

dankzegriefer, 2016-08-26:
@Margen67 Hm? But OP was describing install for unofficial sysmodules? When was this added?

Mrrraou, 2016-08-26:
TwlBg is not a sysmodule.

dankzegriefer, 2016-08-26:
@Mrrraou It isn't? I thought it was a TWL SysModule.

AuroraWright, 2016-08-26:
6.1.1 makes it selectable

closed 2016-08-26 by AuroraWright

noX1609, 2016-08-29:
Hi,

i have the same Problem with Version 6.1 and 6.1.1
Haven't selected the TwlBg in Config so it should be off but the card doesn't work with 6.1.1
If i go back to 6.0.1 it works without a problem but there is the problem with the Settings that sometimes on emuNAND it shows sysNAND and in sysNAND it shows emuNAND. This is fixed in 6.1.1 but as i said the DSTwo Plus doesn't work with 6.1.1 with the patch disabled.

irpacynot, 2017-04-08:
Having a similar issue regarding DSTWO Plus. The system (O3DS) just shuts off when trying to load the card from the Home menu. I tried installing the two cias as per the trouble-shooting FAQ, but it didn't help. I'm on Luma3DS 6.6.

Margen67, 2017-04-08:
@irpacynot https://3ds.guide/troubleshooting

irpacynot, 2017-04-08:
Thanks. Unfortunately, none of those links work. I would otherwise try to install all of those cias.

irpacynot, 2017-04-08:
Never mind. I didn't realize they were torrent files. But I downloaded all of the files and tried importing them, but each one said "program already exists, import cancelled." Tried to load the flash card, but it still auto-shutdown the system.

Margen67, 2017-04-09:
What entrypoint are you using? (A9LH, menuhax, etc) Are you using emuNAND?