LumaTeam/Luma3DS #171: AGB Boot option ignored.

Autz64, 2016-08-26:
I'm using the latest version of Luma3DS (No-Dev) and i checked the option **Show GBA boot screen in patched AGB_FIRM**. However, when i boot a GBA game the boot animation doesn't show and loads directly to the game. I erased the config.bin and setted the options again and still doesn't show the animations. 

I don't know if is relevant here, but i have the no-screeninit version of arm9loaderhax.

Margen67, 2016-08-26:
Have you installed a patched AGB_FIRM cia?

Autz64, 2016-08-26:
No. I can play GBA VC games normally but i can't see the intro.

cheatfreak47, 2016-08-26:
Try reinstalling AGB FIRM. @Autz64

Autz64, 2016-08-26:
I installed a clean AGB_FIRM and now the boot animation appears. But is pretty odd, i don't remember having installed a patched AGB_FIRM in the past.

closed 2016-08-26 by Autz64