LumaTeam/Luma3DS #173: Crash if no option is selected in the configuration menu

BernardoGiordano, 2016-08-26:
Tried 6.1 on a CFW 11.0 sysnand, no emunand. I felt like I didn't have to select any option, because I don't have an emunand, so I thought Autoboot Sysnand was superfluous. When I pressed start to save, it remained in that screen, forcing me to shut down.

TuxSH, 2016-08-26:
I deleted my configuration file, and pressed start with no option on. Didn't freeze.

BernardoGiordano, 2016-08-26:
Just tried. I deleted my config.bin . Crash still happens.

Could I add something? I think that whenever a update changes something related to configuration managemente, the cfw should delete the config.bin on first launch to clean everything. I moved from a past version, and I had the first 3 option checked. When I opened the configuration menu on the new update, there were things that I did not select in the past version, like the ones that boot the second emunand instead of the first. Obviously, the stuff I did not select were at the same place of the 3 option that I selected time ago.

BernardoGiordano, 2016-08-26:
Ah, other thing. When this happens and i forcedly shutdown, when I reboot I don't go directly to the config menu (it seems that a config.bin file is still created and found on the SDcard, I assume), but there is no screen init at all. Also a minute later, the screens do not init and obviously the system doesn't start at all. I need to reboot another time, pressing SELECT and select Autoboot Sysnand to get everyhting okay

TuxSH, 2016-08-26:
You're not keeping START held down, are you?

BernardoGiordano, 2016-08-26:
Nope.. just pressed for a moment. I can make a little video about this issue

TuxSH, 2016-08-26:
Try the dev branch to see.

BernardoGiordano, 2016-08-26:
It stucked for a while, and then this appeared: 
[image]

BernardoGiordano, 2016-08-26:
When I try to reboot instead, I have other values in the registers as this pic says: 
[image]

BernardoGiordano, 2016-08-26:
When I reboot, I have other values in the registers. It goes immediately here: 
[image]

AuroraWright, 2016-08-26:
Try again on 6.1.1

TuxSH, 2016-08-26:
Can be a possible duplicate of #169 as well, if you used to have an emunand.

BernardoGiordano, 2016-08-27:
The crash still happens on 6.1.1 . Reading issue #169 I assume this problem is due to a past GW emuNAND (that I used to have before a redNAND). I formatted my SD Card with the 'No emuNAND' option on EmuNAND9. I assume it was a quick format and not a full one

TuxSH, 2016-08-27:
Yeah, that's probably the issue. Try a full format.

BernardoGiordano, 2016-09-01:
Doing the full format between 24hours, I think. Will let you know

BernardoGiordano, 2016-09-04:
Full format done. Issue still persists.  Currently running 6.1.1-7fb17ce5

Margen67, 2016-09-04:
Try looking at the SD card in Disk management in Windows

BernardoGiordano, 2016-09-04:
I made the full format with that tool on Windows

AuroraWright, 2016-09-04:
The issue is surely with your SD card, as I tested the same thing (clean SD, copied the .bin and pressed start on boot), works fine. Maybe windows's tool didn't erase the ncsd header for some reason, if you can do a dd with zeroes from linux

closed 2016-09-04 by AuroraWright