LumaTeam/Luma3DS #174: Sleep mode download not work

celula82, 2016-08-26:
In 6.0, 6.0.1 and 6.1, sleep mode download not work, the console ask update, but I have 11.0 arm9. In 5.5 works fine.

[image]

thejsa, 2016-08-26:
I can confirm this. Trying to use Sleep Mode Download does, in fact, yield a System Update message ("In order to use this function, you must perform a System Update." etc)

TuxSH, 2016-08-26:
Hm, this is a known bug, try switching the option "use sysnand firm if booting with R"

closed 2016-08-26 by TuxSH

celula82, 2016-08-26:
The use sysnand firm if booting with R option dont fix this. Only works in 5.5 with sysNAND is updated option. Why close this if it is not fix?

AuroraWright, 2016-08-26:
Likely fixed in 6.1.1

Margen67, 2016-08-27:
Is there a firmware.bin in the luma folder? If so, try deleting it.

celula82, 2016-08-27:
No firmware.bin in luma folder. Error is in 6.1.1 too, but only en my new normal. I have a XL version and with 6.1.1 its works fine OK.