LumaTeam/Luma3DS #175: GBA VC Saving Stops after First Save.

CaelThunderwing, 2016-08-26:
(yes to dispel any common questions Yes, i use a bootmanager, Yes Luma's path is set No I'm not using a Dev version, Yes I'm using a SysNand setup [A9LH], and i have an O3DS)

i don't remember how far back this bug started to occur but my GBA VC's seemed to stop saving _**AFTER**_ the first time they make thier save. I thought i somehow messed up when this first occurred when trying AGBSAVE9, so i deleted the .bin it created and tried again.  To give an example:

A custom VC made of Kingdom Hearts CoM, I'd save in the first town in Castle Oblivion (Traverse town) at the first save card i can obtain. wait a few press Home and return to Home menu (which ends up loading ctrbootmanager9 then luma). Reload the VC .. save's still there.  progress in game some , save and home-> Luma-> VC again (be it if i load it at a later time or another day) this time the progress is GONE, as if it never saved or didn't return to sysnand to not loose progress. 

It wasn't always like this and had been playing my GBA VC's made w/ the VC Tool for awhile like this, i believe it started in one of the 5.XX versions of Luma.

cheatfreak47, 2016-08-26:
Seems to be a user specific issue, I have 49 GBA games and all of them are saving fine on my end.

CaelThunderwing, 2016-08-26:
@cheatfreak47 are you loading luma first as arm9loaderhax.bin then chainloading your A9 binaruies as needed or in teh same boat a Bootmanager-> Luma if not would make a small diff and could be the only thing atleast i can see why im having these issues (using a Bootmanager over letting Luma load the rest for me

AuroraWright, 2016-08-26:
6.1.1 should fix this

closed 2016-08-28 by TuxSH