LumaTeam/Luma3DS #176: Permanent 300 play coins

poutros, 2016-08-26:


CoolFangs, 2016-08-27:
But you can already just use JKSM to set it to 300.

Mrrraou, 2016-08-27:
There is no need for that in Luma. Some people want to earn those normally, without cheating; and such a patch is pointless.

poutros, 2016-08-27:
@CoolFangs And do everytime? It's pretty tedious you know.

@Mrrraou Why not put it as an option?

Mrrraou, 2016-08-27:
Because Luma's purpose is not cheating. Just use JKSM for that. Doing such a patch is not worth it.

poutros, 2016-08-27:
@Mrrraou The only real reason for the twlbg patches on the non-dev version is to run nitrohax, a cheating tool....

cheatfreak47, 2016-08-27:
@poutros There's homebrew benefits and potential benefits in the area of running DS games from the SD card as well. It's a useful feature. 

300 playcoins is a neat idea but what the heck are you doing where you blow all 300 playcoins so fast that it's annoying to just open JKSM or something to give youself more coins? Is it that big of a hassle? Is this really something that needs added at a CFW level? I don't think so.

poutros, 2016-08-27:
@cheatfreak47 First, that's not the current reason why it's sought out after, is it? Maybe for *devs* it is but not normal users....
Games that I personally use play coins in:
Streetpass mii plaza
Kirby planet robobot
Super pokemon rumble
The Legend of Zelda: A Link Between Worlds
Kid Icarus: Uprising

Also, thsi:
http://nintendo.wikia.com/wiki/Play_Coins

cheatfreak47, 2016-08-28:
@poutros Real Nintendo 3DS devs have to use an SDK tool called Play Coin Setter, to... guess what? Set their play coin amount manually with a menu.

Homebrew already has even better options for this actually.
[Gimmie 300 Coins - CIA that you run, it sets it and quits, super fast](https://gbatemp.net/threads/release-gimme-300-coins-the-fastest-way-you-will-get-300-coins-ever.421461/)
[Play Coin Setter - 3dsx Homebrew](https://gbatemp.net/threads/homebrew-play-coin-setter-ninjhax-3dsx-edition.377452/)
[For Anyone Who Walks A Lot - a tool to bypass the 10 coin limit](https://github.com/iamevn/for-anyone-who-walks-a-lot)

The mere fact that "For Anyone Who Walks A Lot" exists means some people want to earn their play coins either legit or semi-legit. 

A literal **cheat code** is not getting added directly into Luma3DS when there are other options available, just to save you 30 seconds every 3-8 weeks that most people run out of playcoins. If you run a tap of 300 play coins out any faster than that, you're the problem.

pbanj, 2016-08-28:
as someone who uses playcoins a lot and uses jksm to get them, this is a dumb idea. it's not hard to get 300 play coins with jksm or the other tools.

closed 2016-08-28 by TuxSH

poutros, 2016-08-28:
@cheatfreak47 I think that the fact that the community has made so many apps related to one single problem speaks volumes of said problem.