LumaTeam/Luma3DS #178: NTR modd doesn't work

trbocode, 2016-08-27:
Today I updated my cfw to 6.1.1 and NTR mode stopped working.
I use Luma dev. Tried disabling twlbg, still black screen.
6.1 works perfectly for that matter.
I use a dstwo if that matters.

james-d-elliott, 2016-08-27:
Seems to be working via my n3ds, launch bootntr without the option on and seems fine.

trbocode, 2016-08-27:
NTR=DS games, not the CFW... 
Thought I was clear with me mentioning twlbg and dstwo...

james-d-elliott, 2016-08-27:
Ah, should have made sense, just personally never heard of dstwo flashcarts so I missed it. I'm able to run NTR games/carts at present on my n3ds. Also not discounting your issue, since you've already established it works with 6.1. Just trying to help narrow it down. c;

trbocode, 2016-08-27:
I also tried to run the dsi homebrew launcher (Blue icon) With no avail. I will try the non-dev version and report.

trbocode, 2016-08-27:
The non dev version seems to run fine. While the dev version, even with disabled errdisp, doesn't work

TuxSH, 2016-08-27:
Fixed in https://github.com/AuroraWright/Luma3DS/commit/dca612ffd76a5380af3d8c1601b3e24a3e5069ae

closed 2016-08-27 by TuxSH