LumaTeam/Luma3DS #179: Luma and 6.x?

gunner007cd, 2016-08-27:
Dusted off an old 2DS I had from back in the day. Went through the whole A9LH process. A9LH works fine and chain loading payloads works just fine. However, booting to Luma just hangs on the splash screen. With the SD out, the system boots normally to 6.0.0-12U.

Is Luma able to launch a sysNAND that low? I know the guide mentioned 4.0-4.5 require firmware.bin but for the life of me can't figure out why it freezes.

gunner007cd, 2016-08-27:
Assuming it's probably 6.x, used a 9.2 CTRNAND autoxfer and was able to boot fine.

closed 2016-08-27 by gunner007cd

TuxSH, 2016-08-27:
I know where the issue is: in the reboot patches. The below file should dump a file named `/luma/process9.bin` when trying to boot, could you try it and send that file to me?

[arm9loaderhax.zip](https://github.com/AuroraWright/Luma3DS/files/440581/arm9loaderhax.zip)

TuxSH, 2016-08-27:
Fixed in a5c6b908b6d42cc2371710a73805041963766b4a

gunner007cd, 2016-08-27:
I reverted back with a NAND dump, fixed as you said. THX!