LumaTeam/Luma3DS #181: [BUG] Home Menu Boots Slower with 6.1.1

SirBeethoven, 2016-08-27:
Thanks for the new update! Although, it seems that my N3DS actually boots up the homemenu slower:  
9.88 before  
11.41 after   

It seems that another user is having this problem too [here](https://m.reddit.com/r/3dshacks/comments/4zrjb1/luma3ds_v611_released_with_a_125_second_speed/d6yhqij?context=3)

TuxSH, 2016-08-27:
Weird. But at least it should be consistent with the official firmware.

closed 2016-08-27 by TuxSH

SirBeethoven, 2016-08-28:
What do you mean? Are you seriously closing this? All due respects dude, are you seriously not going to offer any help?

By the way, the latest Luma boots to a black screen (this is not the updates fault)

cheatfreak47, 2016-08-28:
@SirBeethoven Star Updater is likely responsible for that. Updating via that caused black screens, it should be fixed now, I think. Luma 6.1.1 when manually installed is fine. 

I can't vouch for the difference, It seems speedier to me, and I have a 64GB SDXC card formatted with 32KB Cluster Size.

What are the details of your SD Card?

SirBeethoven, 2016-08-28:
64GB 32KB Cluster. N3DS A9LH SysNAND

None of the manual works with this either: http://astronautlevel2.github.io/Luma3DS/

SirBeethoven, 2016-08-28:
Oh, and Sandisk Ultra XC I with the 10 circled

cheatfreak47, 2016-08-28:
@SirBeethoven His hourlies are why Star Updater broke. It's still not fixed? I thought he fixed it already. 
¯\_(ツ)_/¯

SirBeethoven, 2016-08-28:
Oh! Hmm idk. Hopefully this whole boot speed will be resolved.

cheatfreak47, 2016-08-28:
Personally I can't reproduce what you're seeing, boot speed seems to be faster in my tests.

cheatfreak47, 2016-08-28:
Well, tell you what, I'll do some properly controlled tests and post the results.

*6 tests later with a stopwatch*

6.1.1 Results:
10.69s
10.36s
10.51s
Average: 10.52s

6.1 Results 
10.13s
9.99s
10.00s
Average: 10.04s

Well I'll be damned, Seem's you may be onto something @SirBeethoven but, .48 seconds probably isn't worth reopening the issue over. @TuxSH What do you think?

SirBeethoven, 2016-08-28:
Thank you for your help cheatfreak4! :)

TuxSH, 2016-08-28:
Given that 1) it's faster for many people and 2) this is closer to official firmware behavior, I won't change it.

SirBeethoven, 2016-08-28:
Ok, thanks Tux. I guess I'll deal with it. It is probably just my larger SD card

SirBeethoven, 2016-08-28:
But, if there's a way to detect if it's a N3DS with a larger SD card and flip it to the old setup

Or, if it was able to be selectable, that would ROCK! :) Thank you so much!