LumaTeam/Luma3DS #184: DSiWare titles don't boot anymore after enabling and disabling TwlBg patches

smileyhead, 2016-08-28:
I wanted to test out this new feature, and after testing and disabling it, DSiWare titles don't boot anymore. (They didn't boot with the setting on, either.)

poutros, 2016-08-28:
Zelda 4 swords anniversary edition boots with either option here.

smileyhead, 2016-08-28:
It doesn't boot for me. :-/

poutros, 2016-08-28:
Try reinstalling a clean twl firm. https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-Troubleshooting#i-have-issues-with-gbadsdsiware-games-and-i-have-installed-a-patched-agbtwl-firm-cia-in-the-past

smileyhead, 2016-08-28:
I didn't install a modified one in the first place.

poutros, 2016-08-28:
Try reinstalling it either way. It won't do any harm if you properly follow the instructions.

TuxSH, 2016-08-28:
[out.zip](https://github.com/AuroraWright/Luma3DS/files/441129/out.zip)

smileyhead, 2016-08-28:
What's this?

TuxSH, 2016-08-28:
My build of the lastest commit, which should work.

closed 2016-08-28 by TuxSH