LumaTeam/Luma3DS #185: Complex PIN combination

adven142, 2016-08-28:
Use the directional pad, shoulder buttons and face buttons to create a PIN up to 32/or higher characters in length.

cheatfreak47, 2016-08-28:
wha- why would you want a pin anywhere near that long?

astronautlevel2, 2016-08-28:
Do the konami code 3 times

Favna, 2016-08-29:
32 seems overkill but I was actually going to ask sort of the same as a future request where we can end the pin length at any time by pressing the Start button and can use Up/Down/Left/Right/ABXY. The reason for ME personally is well... [the single only thing Konami is still good for - Konami code](https://en.wikipedia.org/wiki/Konami_Code)

I seriously want my pin to be up up down down left right left right A B. I got that so memorized.

hartmannaf, 2016-09-02:
If you only need the pin to prevent the usage of other payloads than luma (thinks like decrypt9 and other payloads that could cause damage), you could simply use BootCTR9. 
It supports the usage of more keys and longer combinations.

Favna, 2016-09-02:
> If you only need the pin to prevent the usage of other payloads than luma (thinks like decrypt9 and other payloads that could cause damage), you could simply use BootCTR9. 
It supports the usage of more keys and longer combinations.

Nah that is not at all why I want a pin. I want it for my device in general.

AuroraWright, 2016-09-04:
v6.1.2 supports the directional pad and up to 8 digits, which should allow for a very large range of combinations :)

closed 2016-09-04 by AuroraWright