LumaTeam/Luma3DS #187: Latest update 6.1.1 not booting into sysnand

GGBEASTBOI, 2016-08-30:
Hi guys, 
  I updated to the latest luma3ds yesterday and when I reboot and it went back to ver instead of sys, I assume that's not in the sysnand. 

  After a few research I update my luma3ds with the hourly release that fixed the problem, it boots into the sysnand again. Maybe some people have the same issue, just want to post here to let those people know. Sorry this might not be in the right section

astronautlevel2, 2016-08-31:
Hold select when booting and make sure "Show current NAND in settings" is enabled.

GGBEASTBOI, 2016-08-31:
Ya all of the options (4 of them ) are selected, but it didn't work, so I update from the hourly one it works now.

GGBEASTBOI, 2016-08-31:
Maybe just for my n3ds xl not sure

AuroraWright, 2016-09-04:
v6.1.1 fixed this

closed 2016-09-04 by AuroraWright