LumaTeam/Luma3DS #188: 2 black box in the home menu

JustForNothing, 2016-08-31:
I have these black box since luma 5.5....
I don't have this problem with Gateway...
[image]

TuxSH, 2016-08-31:
It's probably titles installed to NAND by error using an old version of FBI.

Install the newest CIA version of FBI (to SD), and you shouldn't worry about having new black boxes like that one. Then, to make the current ones dissapear, provided you rembember what you installed, delete the corresponding titles.

fox8091, 2016-09-01:
This could also be an issue due to downgrading. If the title was not originally on this firmware, it has a chance to show up on the Home Menu.

JustForNothing, 2016-09-02:
I have already the lastest version of FBI, and I think the title are from system file, but I don't remember which files

AuroraWright, 2016-09-04:
Not a Luma problem anyway.

closed 2016-09-04 by AuroraWright