LumaTeam/Luma3DS #189: DSi Ware/DS Games give me a black screen

fm360, 2016-08-31:
Problem is in the title
N3DS 9.2 Sysnand 11.0 REDNand  Menuhax
I have never installed any TWL/AGB patches in the past for anyone that is wondering.

Margen67, 2016-09-01:
>Menuhax REDNand

Well there's your problem

fm360, 2016-09-01:
Whats wrong with Menuhax and REDNand? Im kind of a noob

Margen67, 2016-09-01:
Waste of SD card space, slow, low boot rates, can't use updated SysNAND, can't boot CFW without an SD card and leaves the system vulnerable to an update, can't unbrick without a hardmod, needs DSiWare/GBA to be installed on both Emu and SysNAND, etc.

fm360, 2016-09-01:
can you show me a good tutorial for installing A9LH and also where can I find the CIA files for the DSiWare and GBA stuff

Margen67, 2016-09-01:
https://github.com/Plailect/Guide/wiki

TuxSH, 2016-09-03:
For the issue itself I lack many details...

closed 2016-09-03 by TuxSH