LumaTeam/Luma3DS #192: TWLbg patch still needed for a lot of people

trbocode, 2016-09-04:
A big majority of the patched twlbg users use flashcart fowarders. Now they can't use them anymore.

AuroraWright, 2016-09-04:
Only affects DSTWO "forwarders", and the patch itself is not needed anymore since you can now switch from TWL to NTR mode thanks to @Apache_Thunder's work. Doing so would even make a lot more games work which now fail due the twbg patch and AP checks.

closed 2016-09-04 by AuroraWright

trbocode, 2016-09-04:
On the DSTWO, it works fine on all games. Including Pokemon.

ApacheThunder, 2016-09-04:
Some of the forwarders use an older fork of my NTR Launcher. NTR Launcher has since been updated to operate from TWL mode. It's just a matter of time before the forwarders get updated to do the mode switch too. Users who still have to use the twlBg patch can remain on a version of Luma that still has it.

Perhaps keep the patch for the dev build, but for regular Luma, I don't think it will be needed for much longer.

trbocode, 2016-09-04:
What about arm9 OC? Really useful in some games (Trackmania for example)

ApacheThunder, 2016-09-04:
Well you can simply modify my NTR Launcher to not downlock before it switches into NTR mode and locks SCFG. The SCFG_CLK register controls that. Disable the code for that in both arm7 and arm9 and it will use TWL clock speeds in NTR mode. ;)