LumaTeam/Luma3DS #193: Can't compile

XTWebdesign, 2016-09-04:
When i try to compile i get some code errors

```
C:\cygwin64\home\Lilligant\Luma3DS>make
ARMIPS Assembler v0.8.0 (Jan 25 2016 00:21:31) by Kingcom
Done.
ARMIPS Assembler v0.8.0 (Jan 25 2016 00:21:31) by Kingcom
Done.
ARMIPS Assembler v0.8.0 (Jan 25 2016 00:21:31) by Kingcom
Done.
make[1]: Entering directory `/c/cygwin64/home/Lilligant/Luma3DS/injector'
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro//libctru/include -DARM11 -D_3DS   -c -o build/fsldr.o source/fsldr.c
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro//libctru/include -DARM11 -D_3DS   -c -o build/fsreg.o source/fsreg.c
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro//libctru/include -DARM11 -D_3DS   -c -o build/ifile.o source/ifile.c
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro//libctru/include -DARM11 -D_3DS   -c -o build/loader.o source/loader.c
source/loader.c: In function 'loader_LoadProcess':
source/loader.c:198:3: error: unknown type name 'CodeSetInfo'
   CodeSetInfo codesetinfo;
   ^
source/loader.c:247:24: error: request for member 'name' in something not a structure or union
     memcpy(&codesetinfo.name, g_exheader.codesetinfo.name, 8);
                        ^
source/loader.c:248:16: error: request for member 'program_id' in something not a structure or union
     codesetinfo.program_id = progid;
                ^
source/loader.c:249:16: error: request for member 'text_addr' in something not a structure or union
     codesetinfo.text_addr = vaddr.text_addr;
                ^
source/loader.c:250:16: error: request for member 'text_size' in something not a structure or union
     codesetinfo.text_size = vaddr.text_size;
                ^
source/loader.c:251:16: error: request for member 'text_size_total' in something not a structure or union
     codesetinfo.text_size_total = vaddr.text_size;
                ^
source/loader.c:252:16: error: request for member 'ro_addr' in something not a structure or union
     codesetinfo.ro_addr = vaddr.ro_addr;
                ^
source/loader.c:253:16: error: request for member 'ro_size' in something not a structure or union
     codesetinfo.ro_size = vaddr.ro_size;
                ^
source/loader.c:254:16: error: request for member 'ro_size_total' in something not a structure or union
     codesetinfo.ro_size_total = vaddr.ro_size;
                ^
source/loader.c:255:16: error: request for member 'rw_addr' in something not a structure or union
     codesetinfo.rw_addr = vaddr.data_addr;
                ^
source/loader.c:256:16: error: request for member 'rw_size' in something not a structure or union
     codesetinfo.rw_size = vaddr.data_size;
                ^
source/loader.c:257:16: error: request for member 'rw_size_total' in something not a structure or union
     codesetinfo.rw_size_total = data_mem_size;
                ^
source/loader.c:258:11: warning: implicit declaration of function 'svcCreateCodeSet' [-Wimplicit-function-declaration]
     res = svcCreateCodeSet(&codeset, &codesetinfo, (void *)shared_addr.text_addr, (void *)shared_addr.ro_addr, (void *)shared_addr.data_addr);
           ^
source/loader.c:261:13: warning: implicit declaration of function 'svcCreateProcess' [-Wimplicit-function-declaration]
       res = svcCreateProcess(process, codeset, g_exheader.arm11kernelcaps.descriptors, count);
             ^
make[1]: *** [build/loader.o] Error 1
make[1]: Leaving directory `/c/cygwin64/home/Lilligant/Luma3DS/injector'
make: *** [build/injector.h] Error 2

C:\cygwin64\home\Lilligant\Luma3DS>
```

TuxSH, 2016-09-04:
Your ctrulib installation is outdated, update ctrulib to the latest commit.

closed 2016-09-04 by AuroraWright

XTWebdesign, 2016-09-04:
You sure it's outdated?
```
[msys]
Version=1.0.17
[devkitARM]
Version=45
[devkitPPC]
Version=27
[devkitPSP]
Version=16-1
[libgba]
Version=20090222
[libgbafat]
Version=1.0.14
[maxmodgba]
Version=1.0.9
[libnds]
Version=1.5.12
[libndsfat]
Version=1.0.14
[defaultarm7]
Version=0.6.0
[filesystem]
Version=0.9.12
[dswifi]
Version=0.3.17
[libmirko]
Version=0.9.7
[maxmodds]
Version=1.0.9
[libctru]
Version=1.1.0
[citro3d]
Version=1.1.0
[libogc]
Version=1.8.12
[libogcfat]
Version=1.0.14
[3dsexamples]
Version=20160106
[ndsexamples]
Version=20140401
[gbaexamples]
Version=20090222
[gp32examples]
Version=20051021
[cubeexamples]
Version=20110620
[wiiexamples]
Version=20110620
[pspdoc]
Version=20051113
[gcube]
Version=0.4.0
[insight]
Version=7.3.50.20110803
[pnotepad]
Version=2.0.8.718
```

usineur, 2016-09-04:
sure, you need latest git revision. The version number is unchanged for a while.