LumaTeam/Luma3DS #194: 6.1.2 black screen, does not boot into home menu

vloeibaarglas, 2016-09-04:
Issue experienced here:
https://www.reddit.com/r/3dshacks/comments/514712/luma3ds_v612_released_bugfixes_etc/d79b34h

Deleted config from /luma. Tried with only Autoboot Sysnand check, tried with GBA splash and Autoboot Sysnand checked. Chainloader and settings menu works fine.

AuroraWright, 2016-09-04:
Did you copy the official .bin file on the root or used some automated tool to update? If so download the release from here and try again. It works on my New3DS with 11.0

vloeibaarglas, 2016-09-04:
Manual copy. Was on 6.0.1 until a while ago.  6.1.1 works fine (copy to SD, delete config). 6.1.2 doesn't with same procedure.

AuroraWright, 2016-09-04:
Try the dev 6.1.2 version and provide the exception dump from the /luma/dumps folder

vloeibaarglas, 2016-09-04:
See attached.

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/454104/crash_dump_00000000.zip)

Gunblade08, 2016-09-04:
I am having an issue and was wondering if someone can help.When I download Luma, the arm9loaderhax.bin is nowhere to be found where can i find it?

Favna, 2016-09-04:
Also happens here. Used the 6.1.2 version downloaded directly from the releases. Attached a zip with the crashdump using the (from releases) dev version tried afterwards. Config for regular was 

- screen brightness: 4x (i got no screen-init anyway)
- pin-lock: none (i cri over 8 and not 10. muh konami code ;-;)
- CPU clock + L2
- Autoboot sysnand
- Use sysnand FIRM
- display NAND

Model device: New3DS XL EUR
Firmware: 11.0.0-33 (clean, no system mods)

***

Oh and aside from the file here is the file already parsed with the python script included:
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/454120/crash_dump_00000000.zip)

```
exception_dump_parser.py crash_dump_00000000.dmp
Processor: ARM11 (core 1)
Exception type: data abort
Current process: loader   (0000000000000000)

Register dump:

r0             10000000            r1             1400300c
r2             1400300f            r3             00000007
r4             00000000            r5             14003013
r6             00241ff9            r7             10241ff9
r8             00241ff9            r9             10241ff9
r10            fffffff0            r11            1ff83000
r12            10241ffb            sp             0ffff9f0
lr             00000055            pc             14000a3c

cpsr           80000010            dfsr           00000007
ifsr           00001009            far            10242000
fpexc          00000000            fpinst         ee000a00
fpinst2        b4303189

Code dump:

14000A10:  09 00 00 ea 01 e0 d2 e4  01 40 fc e5 0e 00 54 e1   |.........@....T.|
14000A20:  05 00 00 1a 05 00 52 e1  f9 ff ff 1a 09 00 a0 e1   |......R.........|
14000A30:  01 db 8d e2 04 d0 8d e2  f0 83 bd e8 03 20 d7 e7   |............. ..|

Stack dump:

0FFFF9F0:  06 00 00 00 03 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA00:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA10:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA20:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA30:  02 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA40:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA50:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA60:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA70:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA80:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFA90:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFAA0:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFAB0:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFAC0:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFAD0:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFAE0:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFAF0:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFB00:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFB10:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFB20:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFB30:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFB40:  08 00 00 00 05 00 00 00  08 00 00 00 08 00 00 00   |................|
0FFFFB50:  08 00 00 00 08 00 00 00  08 00 00 00 08 00 00 00   |................|
[... 50 lines cut ...]
0FFFFE80:  00 20 24 00 00 00 00 00  78 ff ff 0f 84 30 f8 1f   |. $.....x....0..|
0FFFFE90:  00 30 f8 1f 84 30 f8 1f  00 20 24 00 d4 70 01 00   |.0...0... $..p..|
0FFFFEA0:  1c 17 04 00 20 ff ff 0f  01 00 00 00 00 00 00 00   |.... ...........|
0FFFFEB0:  00 00 00 00 00 00 00 00  00 00 00 10 00 00 00 00   |................|
0FFFFEC0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0FFFFED0:  00 00 00 00 00 00 00 00  02 00 00 00 07 00 04 00   |................|
0FFFFEE0:  34 01 00 00 01 00 00 00  88 3f 15 00 00 00 00 00   |4........?......|
0FFFFEF0:  88 3f 15 00 00 00 00 00  00 00 00 00 00 00 00 00   |.?..............|
0FFFFF00:  08 00 13 00 00 00 00 00  08 80 13 00 00 00 00 00   |................|
0FFFFF10:  88 3f 15 00 00 00 00 00  88 3f 15 00 00 00 00 00   |.?.......?......|
0FFFFF20:  09 80 12 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0FFFFF30:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 10   |................|
0FFFFF40:  06 02 00 00 00 60 20 10  24 00 00 00 00 a0 22 10   |.....` .$.....".|
0FFFFF50:  18 00 00 00 42 02 00 00  00 00 10 00 06 02 00 00   |....B...........|
0FFFFF60:  00 60 30 00 24 00 00 00  00 a0 32 00 18 00 00 00   |.`0.$.....2.....|
0FFFFF70:  42 02 00 00 00 00 00 00  00 00 00 10 30 00 04 00   |B...........0...|
0FFFFF80:  00 ff ff 0f 34 ff ff 0f  00 00 10 00 31 00 00 00   |....4.......1...|
0FFFFF90:  00 10 13 00 09 00 00 00  00 a0 13 00 04 00 00 00   |................|
0FFFFFA0:  31 00 00 00 09 00 00 00  07 00 00 00 68 44 00 14   |1...........hD..|
0FFFFFB0:  02 8a 00 00 30 00 04 00  00 00 00 00 01 00 00 00   |....0...........|
0FFFFFC0:  02 13 00 00 30 01 04 00  00 00 00 00 00 00 00 00   |....0...........|
0FFFFFD0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0FFFFFE0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0FFFFFF0:  00 00 00 00 00 00 00 00  00 00 00 00 a8 02 00 14   |................|
```

exegg, 2016-09-04:
This happened to me with 6.1.1. I downloaded the official release directly from here. Deleted config.bin, checked the right options, booted my payloads, but trying SysNAND was a black screen everytime... Until I took out the SD card, pulled it back in and then it worked again. Could this be related to the faster boot implemented recently?

AuroraWright, 2016-09-04:
Does this work?

[arm9loaderhax.bin.zip](https://github.com/AuroraWright/Luma3DS/files/454172/arm9loaderhax.bin.zip)

vloeibaarglas, 2016-09-05:
Yup, the dev version boots. I didn't test any features, but I assume everything is working. Looks like you fixed it.

AuroraWright, 2016-09-05:
I don't know why though.
Could you also try this?

[arm9loaderhax.bin.zip](https://github.com/AuroraWright/Luma3DS/files/454182/arm9loaderhax.bin.zip)

vloeibaarglas, 2016-09-05:
Second upload also works.

AuroraWright, 2016-09-05:
Round three (sorry about this, I can't reproduce it on my console). This **should** crash I think...
[arm9loaderhax.bin3.zip](https://github.com/AuroraWright/Luma3DS/files/454229/arm9loaderhax.bin3.zip)

vloeibaarglas, 2016-09-05:
Third one does crash. In case you still need the dump, but it seems you have figured out the cause.

[crash_dump_00000002.zip](https://github.com/AuroraWright/Luma3DS/files/454301/crash_dump_00000002.zip)

Gunblade08, 2016-09-05:
ok one more issue. On part 3 it says to Delete the 3ds folder from the root of your SD card if it exists. but then to Copy DspDump.3dsx to the /3ds/ folder on your SD card a couple steps down which one is it?

MelonGx, 2016-09-05:
All using 6.1.2 DEV.
N3DS A9LH Sys 8.1.0-0J Emu 11.0.0-33J --- Booted properly.
O3DS Menuhax Sys 9.2.0-20J Emu 11.0.0-33J --- Booted properly.
2DS A9LH Sys 11.0.0-33J --- Booted properly.
O3DS-2 A9LH Sys 11.0.0-33J --- Crashed.
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/454426/crash_dump_00000000.zip)

james-d-elliott, 2016-09-05:
@Gunblade08 if you're trying to notify @Plialect about his A9LH guide, this is the issues page for Luma3DS and a completely unrelated issue to what you're describing. You should post a new issue here: https://github.com/Plailect/Guide/issues

AuroraWright, 2016-09-05:
The issue is weird, it seems to be caused by the memory searching algo inside loader (which would be weird as it worked without any issue on the arm9 side for a long time). I also changed the linker script for loader for this release, so let's exclude that at least:

[arm9loaderhax.bin4.zip](https://github.com/AuroraWright/Luma3DS/files/454950/arm9loaderhax.bin4.zip)

jgreenwayread, 2016-09-09:
i'm having what seems to be the same issue, here is my dump from bin4
[crash_dump_00000004.zip](https://github.com/AuroraWright/Luma3DS/files/463281/crash_dump_00000004.zip)

AuroraWright, 2016-09-11:
Fixed

closed 2016-09-11 by AuroraWright