LumaTeam/Luma3DS #196: Sleep mode download not work in any 6.XX version

celula82, 2016-09-05:
rying to use Sleep Mode Download does, in fact, yield a System Update message ("In order to use this function, you must perform a System Update." etc) in all 6.XX version of luma (6.1.2 incl.) and only works in 5.5 in my normal new 3ds. My new 3ds XL works well with luma 6.1.2 and sleep mode download. I have the same archives in te sd cards! Why? Please, help.
Look a video:
https://twitter.com/kyosuke82/status/772785052828725252

poutros, 2016-09-05:
Try deselecting "Use sysnand FIRM if booting with R (A9LH)". If you're using this because of ntr, use BootNTR loader from here: https://github.com/SirBeethoven/BootNTR/releases

celula82, 2016-09-05:
Same with deselecting that option. I dont use ntr.

AuroraWright, 2016-09-05:
"My new 3ds XL works well with luma 6.1.2 and sleep mode download." Go to system settings and update, I think some title might be out of date...

Omegadrien, 2016-09-05:
Same issue with me. My n3ds is already up to date (manually updated with 3DSNUS and sysupdater).
New 3DS A9LH 11.0.0-33E SysNand

celula82, 2016-09-06:
When I try update, the console said it is already updated

exegg, 2016-09-06:
It's working just fine on my O3DS. 11.0.0-33U with Luma 6.1.2

Omegadrien, 2016-09-06:
OK, so to be precise, the download works if the 3ds is in sleep mode, but the "Y Complete Download Now" feature doesn't ("you need to preform a system update"...)

exegg, 2016-09-06:
I did exactly what @celula82 posted in the video. No issues.

celula82, 2016-09-06:
The download dont work any mode, y button or in sleep mode

Committee, 2016-09-06:
N3DS XL here on the latest firmware using latest luma3ds release menuhax entrypoint y button update works, have you tried entering the EShop and check if the error persists?

celula82, 2016-09-06:
Yes, I can donwload games from eshop but not with y button or sleep mode.

Margen67, 2016-09-07:
@Committee
>menuhax
ew

celula82, 2016-09-07:
No menuhax, I use arm9 + luma sysnand 11.0 Eur

piratesephiroth, 2016-09-08:
Sleep mode works. "Download now" with the Y button doesn't.

Omegadrien, 2016-09-08:
OK, so good news! With me, the problem is fixed with 
https://github.com/AuroraWright/Luma3DS/commit/b74dda42a2325030b63f9fe93c44ea0d0bc08a51
@piratesephiroth and @celula82 please try.

piratesephiroth, 2016-09-09:
Yep. Seems to be working fine now.

closed 2016-09-09 by TuxSH