LumaTeam/Luma3DS #198: Stuck on AureiNand 5.1.2 Configuration screen

TheAzure289, 2016-09-05:
Tried to fix an online connectivity error by updating the arm9loaderhax.bin, my 2ds wouldn't boot so I went back to my old bin and now I'm stuck at the config screen. It will not save my choices, just refreshes without moving on.

Margen67, 2016-09-05:
Try the latest bin from https://github.com/AuroraWright/Luma3DS/releases

TheAzure289, 2016-09-05:
ok 6.1.2 popped up but froze when i made my choices. my system firmware is updated to the latest. what choices should be on?

TheAzure289, 2016-09-05:
it is still frozen. i selected, show current nand in system settings and show gba boot screen

poutros, 2016-09-05:
Select autoboot sysnand.

TheAzure289, 2016-09-05:
That did the trick! Thank you!!

fox8091, 2016-09-06:
So..if the issue is fixed, can this be closed? Also, this isn't a forum. Things like this belong somewhere like GBATemp.

closed 2016-09-07 by TuxSH