LumaTeam/Luma3DS #200: MVD Service not accessable

thedevleon, 2016-09-06:
When trying to use the mvd service (with the provided example[example](https://github.com/devkitPro/3ds-examples/tree/master/mvd))

 mvdstdInit() returns 0xd8e06406.

As far as I could gather, this means the service is inaccessible.
Would it be possible to add patches to make this accessible from within homebrew apps?

fox8091, 2016-09-06:
This requires service patches, which are implemented in the dev branch.

thedevleon, 2016-09-06:
Thanks, but running the dev version gives me this:

[image]

Mrrraou, 2016-09-06:
Are you sure this is a Luma bug ? This is a crash in the mvd module.
You should use CIAs and add access for the mvd service/port in your CIA exheaders.

closed 2016-11-06 by TuxSH