LumaTeam/Luma3DS #203: Finalizing setup from plailect (double confirm is it got anything missing)

jasontwk85, 2016-09-07:
[image]

Sorry if i am asking wrong thing or impolite.
1. I wanna ask is it i.m missing something about luma3ds?maybe app or tool? After i finish step from plailect.. cos this is the only thing i got..

2. And i wanna ask how to use rom such as .3ds or .cia? Or can send me a link so i can follow and learn.?

Thanks,
And sorry for inconvenience cause.

ihaveamac, 2016-09-07:
this is not a Luma3DS issue. you asked this already at https://github.com/Plailect/Guide/issues/477.

the icon that is selected is a Luma3DS updater, unrelated to the actual custom firmware.

jasontwk85, 2016-09-08:
Sorry bro..i dunno where should i ask to double confirm..
And sorry for being impolite😇

closed 2016-09-08 by jasontwk85