LumaTeam/Luma3DS #204: Luma not booting

pixeled4life, 2016-09-07:
At first: sorry for my bad english, I'm spanish

I know someone with arm9lh with a 3ds that doesn't boot when having r4 flashcart inserted, I cannot tell a lot of details because it's not my ds but I have replicated the same issue in a 3ds of mine that uses emunand(with menuhax) and this is what I found:

-Only happens when the battery is at 3/4 or below but if so it happens every time

-It only happens on the same console(haven't tested to format and retry/boot luma on the 9.2 sysnand)

Flashcart info:

-R4i(not gold)
-in the console it shows an english spongebob game(I'm spanish, eur)
-It works on every console I've tested(eur dslite, eur dsiXL, usa dsi, eur 2DS on 9.2 and 2 different eur 2ds on 11.0)

If you want me to test some more tell me and I will be pleased to help

Margen67, 2016-09-08:
>menuhax

ew

pixeled4life, 2016-09-08:
@Margen67 yeah I know but it's a 2ds I have for my brother that he doesn't use, I use a9lh but from what I can tell it's a problem with luma because menuhax loads but the emunand doesn't and it happened to someone who had arm9lh

TuxSH, 2016-09-08:
Does it boot w/o that r4 slot?

pixeled4life, 2016-09-08:
@TuxSH Yes, completely fine, I didn't want to fix a problem, I just was reporting a problem

AuroraWright, 2016-09-08:
This is a Nintendo bug, not a Luma bug.

closed 2016-09-08 by AuroraWright

pixeled4life, 2016-09-08:
@AuroraWright I thought it was because the 3ds worked perfectly fine with it before installing luma

AuroraWright, 2016-09-08:
We don't touch the cart-related code at all with our patches, try to clean the contacts of your flashcart with alcohol