LumaTeam/Luma3DS #205: Problem with GBA games on emunand

JustForNothing, 2016-09-08:
When I boot a gba game on my emunand, it reset my home menu config.
All game who was in a folder, are now in the home menu when I shut down my 3ds after a gba game

Margen67, 2016-09-08:
>emunand

ew

JustForNothing, 2016-09-08:
nice description of the problem

ihaveamac, 2016-09-08:
do you happen to have menuhax in emunand? it's an exploit in `SaveData.dat` for 10.6+. if it happens after normally rebooting too, you need to remove menuhax from emunand.

JustForNothing, 2016-09-08:
I never installed menuhax

ihaveamac, 2016-09-08:
well, does it still happen after rebooting, or *only* after using GBA games?

JustForNothing, 2016-09-08:
after using GBA games

Margen67, 2016-09-09:
Why are you using an emunand?

Shadow322, 2016-09-09:
@Margen67 why u forcing them to use a9lh if they dont want to?

Margen67, 2016-09-09:
No reason not to.

cheatfreak47, 2016-09-09:
Okay, sounds like something is rewrapping your GBA games. This can be caused by two things:
One is that your NANDs are linked and when you open SysNAND the install of the GBA game there conflicts with the install in your EmuNAND, causing desynced extdata between both NANDs.

It's also possible that your NANDs are linked and when booting EmuNAND, the EmuNAND can't write to the SD Icon Cache because MenuHax has it write protected, causing perpetual wrapped games.

So, in any case it sounds like your NANDs are linked. How many folders in SD:/Nintendo 3DS/ do you have?

If so, you can solve this by running Tiny Format on your SysNAND and reinstalling whatever entrypoint you are using to get into your EmuNAND. (MenuHax reccomended).

From there, if you wish to mitigate possible future issues as the homebrew community moves away from EmuNANDs entirely, the best course of action is to install Arm9LoaderHax and migrate your EmuNAND into your SysNAND by following [this guide](https://github.com/Plailect/Guide/wiki).

For any additional help please ask somewhere else. This is a tracker for reporting issues with the actual custom firmware, not help sorting out user error related problems.

@Shadow322 Nobody's forcing anyone to do anything here, the OP can do whatever they want.
They _should be_ using Arm9LoaderHax, most people should, but there's literally nothing anyone can do to _force_ anyone to do anything.

The reason @Margen67 is suggesting you set up A9LH is because this issue of gifting, NAND linking, and needing an entrypoint in SysNAND are moot with A9LH because you'll only have to work with one SysNAND, and no EmuNAND at all.

JustForNothing, 2016-09-09:
@Margen67, I use my emunand since 2013, and I don't care about your preference

And thanks @cheatfreak47 for your detailed write up

I have 2 folder in SD:/Nintendo 3DS/
The emunand folder, and the "Private" folder

Margen67, 2016-09-09:
@JustForNothing it's not "preference." A9LH is objectively better.

kitling, 2016-09-09:
@Margen67 Stop trying to push your preference on other people. True, I will agree, I think A9LH is better, but get this: you can use an EmuNAND w/ A9LH as well. Most people just don't. However, I don't push this on anynody. (I don't like Gateway, but I don't go around telling people that the *have* to throw away their card.)

In any case, if someone can't get down to 9.2 or lower, they can't do EmuNAND or CFW *anyways*.

Some people don't want to take the risk of the random soft-bricks and issues that can come about while installing a9lh. Some people are comfortable the way they have it, and don't want to mess about and change everything. Really, after they get into CFW, it's all the same. We actually don't know what entry point they're using to get into CFW/EmuNAND, it's possible that they're *already* using a9lh, just with an EmuNAND.

@JustForNothing I'm sorry that I myself don't have a solution for this. Just do what you want to do. Margen posted a link to this issue with a derogatory comment, and I don't think that's okay.

If you happen to be going back into SysNAND inbetween rebooting after a GBA game and re-entering EmuNAND, that could be the issue. If you can unlink your SysNAND from your EmuNAND, then that issue might be solved?

I'm not an expert with EmuNAND, unlinking, or GBA VC injection, or whatever yb any means. I'm just remembering that the whole gift-box issue could be fixed by unlinking. Take my advice with a grain of salt, and do some research.

Good luck. :D

Margen67, 2016-09-09:
@kitling
>Some people don't want to take the risk of the random soft-bricks and issues that can come about while installing a9lh. Some people are comfortable the way they have it, and don't want to mess about and change everything. Really, after they get into CFW, it's all the same.

The guide has so many failsafes at this point that you would have to either be an idiot, or try to brick.

Settling for inadequacy is a poor excuse.

Clearly it's not "all the same" since updated SysNAND wouldn't have the issue this user is having.

If anything, menuhax is unsafe. You can't unbrick without a hardmod, and the sysNAND is still vulnerable to updates in certain cases, like when you don't boot it or don't have the SD card inserted.
A9LH can now boot without the SD card if you use the latest commit of Aurora's A9LH fork.

cheatfreak47, 2016-09-09:
@JustForNothing

You're welcome, but like I said, **this isn't the right place to get help.**
Please close this issue and move to a different website, like GBATemp or Reddit and ask your questions there.

This is an issues tracker, not a support forum.

JustForNothing, 2016-09-10:
Yeah that true, I thought that was an issue.

But that funny how the 3DS have so many kiddies, that was not like this for the NDS scene in the past :^(

Thanks everyone !

closed 2016-09-10 by JustForNothing