LumaTeam/Luma3DS #206: Post luma3ds config, sysNAND fails to update via sysSettings

creedda, 2016-09-09:
Following the Phailect Guide Here:
https://github.com/Plailect/Guide/wiki

See closed issue on that guide:
https://github.com/Plailect/Guide/issues/475#issuecomment-245542236

Old 3DS [10.x.x origininally] Ver. 2.1.0-4U Now

I downgraded from a 10.x.x firmware on an old 3ds, following all of the instructions in Plailect's guide. The update via the sysSettings after the configuration of luma3ds will just not go through. I have tried to update via several different wireless networks (different houses) and have tried standing next to my router, restarting my router, turning off my firewall, and several other tricks I have found on the web. Nothing seems to work. The update gets to about 60-75% before erroring out every time.

For reference, my system setting show "Ver. 2.1.0-4U" instead of a sys or emu version. I think they are supposed to show a different string thanks to the setting in luma3ds?

Recovery mode is one of the strategies I tried. I get an error when booting into recovery mode:
_An error has occurred: 
SAFE_MODE is not supported on 1.x/2.x FIRM 
Press any button to shutdown_

I tried reformatting and doing the update again. Reformatting did clear out the update data so the progress bar starts from the beginning, but when it gets to the same spot as before it still errors out with the same message. After trying a few different things with the reformatted sysnand I restored my original sysnand data from a nand dump I had taken before the reformat. (to preserve some saves)

Another possibility I thought of today would be to insert a cart and cart update, but sadly my 3ds has a physical problem which is preventing carts from being recognized. (hence why I am doing this) But Maybe that might help someone else who has the same problem?

Based on what I've seen and looking into what FIRM protection actually does, could this be an issue with Luma3DS?

Does anybody know if manually updating to 11.0 through sysupdater on Luma3DS works too? Or does that not work?

Any other suggestions?

Let me know if there is any additional data I can gather that would help you troubleshoot...

Thanks again for putting this software together.

TuxSH, 2016-09-09:
>I think they are supposed to show a different string thanks to the setting in luma3ds? 

Not on 1.x/2.x .

>Recovery mode is one of the strategies I tried. I get an error when booting into recovery mode:

Intented.

Let me investigate the issue anyways.

creedda, 2016-09-11:
Just to add bit more information to rule out some things.

I was able to complete the guide using a New 3DS I had as well with no issues, including this part where I update through the system settings. That rules out my connection to the internet as an issue.

I have been trying continuously to update the old 3DS and I get the same error. I have been trying at different times of the day so its not the servers being down.

I have tried with and without a cart inserted (though my old 3DS cannot register that a cart is inserted in the software so I wasn't expecting much)

I also tried injecting FBI into the HS app but that doesn't seem to work, and neither does the home-brew browser exploit so I am not sure how I would even go about upgrading the firmware through sys updater even if that was the right thing to do.

Any other suggestions would be appreciated. Let me know if I can get any info that would help.

arromdee, 2016-09-23:
The changelog for 6.2 says "Fixed support for pre-9.0 FIRMs. Luma3DS now directly supports >= 5.x FIRMs, as intended".   While 2.1 is not >= 5.x, might something have changed that makes it work?

AuroraWright, 2016-09-23:
No

fl0wzen, 2016-09-24:
I would try to do a cartridge update back to 9.2. So you can give it a new try. Plailect have updatet his guide several times, so it's possible that you now can do it right. For example: I had a few problems, one of themt could only be fixed by rerun the updated guide. In your case, it's hard to find the true failure. If your configuration work right, you should be able to complete the guide. If not, you have a clean 9.2 and can start from scratch. 

In case und can afford it,  buy a bigger sd card, so you can bypass problems of updating. 

You can see witch cartridge have the required update  at this Site: 

http://www.3dsdb.com

Take care of the right regioncode

creedda, 2016-09-25:
Replying just to clarify that this is still an issue as I cannot cart update with this particular 3DS. However, that solution might work for some people.

fl0wzen, 2016-09-27:
Sorry, i missed the part where you wrote that your system dont recognized cartridges :-(

pbanj, 2016-09-28:
@creedda https://github.com/Plailect/Guide/wiki/9.2.0-ctrtransfer try that

creedda, 2016-10-06:
Hey Guys. Just wanted to check in and let you know that I tried what pbanj suggested today and it worked!

So that would be my recommendation if you can't cart update.

Thanks so much for all the assistance.

closed 2016-11-06 by TuxSH